prusa3d/PrusaSlicer #268: Slic3r goes not responding over and over Windows 10 Pro

brotherchris, 2017-04-22:
### Version
_Version of Slic3r used goes here_
1.33.8

_Use `About->About Slic3r` for release versions_
1.338.-prusa3d-win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Pro 64bit
### Behavior
* _Describe the problem_

Slic3r goes not responding constantly. Every other click

* _Steps needed to reproduce the problem_

Happens every time I use it.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[3DBenchy.zip](https://github.com/prusa3d/Slic3r/files/949265/3DBenchy.zip)

[config.zip](https://github.com/prusa3d/Slic3r/files/949266/config.zip)

brotherchris, 2017-04-23:
It appears to be the plater tab that causes the not responding. I did upgrade my intel graphics driver, but no change.

bubnikv, 2017-04-24:
Isn't it the same issue as https://github.com/prusa3d/Slic3r/issues/233 ?
Did you update the INF files before updating the Intel HD Graphics driver?
Is the problem specific to the 1.34.1 Slic3r build, or do you have the same issue with Slic3r Prusa Edition 1.33.8?

brotherchris, 2017-04-24:
I reinstalled the intel drivers and it seems to be working now. It happens in 1.33.8 and 1.34.1. Closing bug.

closed 2017-04-24 by brotherchris

bubnikv, 2017-04-24:
Thanks for heads up. Please note that you will likely need to update the
Intel Graphics driver after each major Windows 10 update.

bubnikv, 2018-02-22:
If you have this issue, would you please re-test with the latest beta build and report the results?  Thank you.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta1

The fix consists of removing the OpenGL glFlush() calls. The glFlush() call was superfluous, but it certainly should not lead to such adverse effects. Removal of glFlush() calls certainly solved the issue on one laptop, so let's hope it will fix all the other issues or at least most of them.

Thanks @roesel for providing a laptop for testing, it was a blessing to us.

brotherchris, 2018-02-24:
Put the fix on, so far so good. It looks like this fixed it. Very much appreciated. I will report back if I see anymore issues.