prusa3d/PrusaSlicer #269: Can't slice this file

tjhawkey, 2017-04-23:
### Version
1.34.1

### Operating system type + version
Win10

### Behavior
Can't slice the "Palm left new supports" file in thing 1453190
[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Hugoz12, 2017-04-23:
Link to thing: http://www.thingiverse.com/thing:1453190/#files
Direct-link to STL: http://www.thingiverse.com/download:2253500
Seems to be a duplicate of model import error, as it's not manifold (924 errors).

valasek, 2017-04-23:
Verified on on the same file using Ubuntu 16.04 and version 1.33.8.

[image]

bubnikv, 2017-04-23:
I am looking into this issue in general. Cura and S3D handle these files
much better as they do not try to fix the model in 3D, but only try to
stitch the 2D slices. The same is applicable to Slic3r.

Also interestingly Microsoft provides a high quality mesh repair function
as a standard library on Windows 10, which Slic3r could make use of.

valasek, 2017-04-23:
Looks like stl is broken. Tried to fix it using https://service.netfabb.com/ but got "FAILED: failed to load mesh (#103)".
So followed this Blender tutorial: https://its.yale.edu/how-to/article-opening-and-repairing-3d-model-stl-file-blender and afterwards it was sliced correctly.

[image]

Maybe an enhancement to warn a user before slicing a broken stl file can be made.

**Proposing to close the issue.**

tjhawkey, 2017-04-23:
Thanks for looking! Interestingly, "MakeItPrintable" reported no problems with the file.
S3D will slice it, but if you try to repair in S3D, it really breaks it.

ntoff, 2017-05-07:
Automatically repairing with windows 10's 3dbuilder also appears to yield a sliceable file

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.