prusa3d/PrusaSlicer #270: adding m600 filament change in Slic3r

robseee, 2017-04-23:
Hi,
It would be nice to have the possibility of adding an M600 command at a specific layer height for filament change somewhere in the Slic3r GUI

Thank you,
Roberto

Sebastianv650, 2017-04-23:
Inserting M600 at a specific layer wouldn't be enough. Usualy you want to have the change at some point where you can't see imperfections in the final print, therefore somewhere inside the infill.
At first this looks quite easy: When the user tells Slic3r to do an M600 at x mm layer height, Slic3r should try first to insert it at some infill point. If this is impossible (no infill), take some inner perimeter point. If there is only an external perimeter, take that.

But maybe there is also some reason to have an M600 in a multi-extruder system? Is it enough to specify the number of the active extruder you want to have the change event, applying the same rules? I don't know..

robseee, 2017-04-23:
Good point,
Having a single extruder system I don't know about multiple extruders

I still think it would be nice to have this feature

Regards,
Roberto

bubnikv, 2017-04-24:
We will have such a feature in Prusa Control, which is a Prusa specific
easy to use Slic3r front end.

bubnikv, 2017-04-24:
Also we have a java script application for that.
http://www.prusaprinters.org/color-print/

MStylesMS, 2018-01-07:
I've tried the multilateral version of color print and it does not seem to trigger the color changes.  It appears to only insert "Tx" where "x" is 0-3.  It does not appear to do the retraction, change, and extrude to purge cycle that is needed to pull the old filament out of the way and properly clear the nozzle.  

Is this supposed to be handled in firmware on the PRUSA printers?  If not, that would be a nice feature to add there.  If it is there, how do I make use of it?  For now I'm just using M600 because I am unsure of the GCODE that should be inserted for my MK2sMM.

bubnikv, 2018-01-07:
>  It appears to only insert "Tx" where "x" is 0-3.

In the next step you are supposed to pipe the G-code through a standalone
MM post processor. Frankly I would have to ask where to download it.

supermerill, 2018-01-08:
@ MStylesMS 

With mmu, Withour gui: open your file in a text editor, search for zXX.XX, insert M600 in the blank line.

i've made some 5-colors print with this (1 M600, 4 colors from mmu)
the m600 will change the filament on the current extruder, so do this only at a layer where only one extruder is active, to be sure.

bubnikv, 2018-08-02:
This is a similar request to https://github.com/prusa3d/Slic3r/issues/628
@vojtechkral  will work on that.

PotatoFi, 2018-10-29:
This would be a very welcome feature in Slic3r Prusa Edition. It is very nice to have in PrusaControl, but PrusaControl hasn't seen a release in about a year now and there is an overextrusion bug that keeps me from using it.

lukasmatena, 2018-10-29:
@PotatoFi The work on this feature is nearly completed and it will be part of the 1.42 release. We will see when it happens, there is still quite a lot of other work, because of major changes in the internal architecture which have to be completed in one go, not to mention testing. But you can look forward to similar behaviour as you know from PrusaControl.

bubnikv, 2018-12-11:
The M600 aka "color print" was implemented in Slic3r PE 1.42 alpha to be released soon. Closing.

closed 2018-12-11 by bubnikv