prusa3d/PrusaSlicer #271: extraneous non-printing moves

jhitesma, 2017-04-23:
### Version
1.34.1-prusa3d

### Operating system type + version
macos 10.12.4

### Behavior
On a part with inner structure and outer structure slic3r is generating a lot of extraneous non-printing moves.  When printing the outside structure it makes moves to the inside structure - then immediately returns to the outside structure without actually printing anything on the inside.  

I caught video of this on the last layer that exhibited this at the 0:15 second point:
https://www.youtube.com/edit?o=U&video_id=dl_tNpNebmo

#### STL/Config (.ZIP) where problem occurs
[example.zip](https://github.com/prusa3d/Slic3r/files/950027/example.zip)

Zip file with .stl, config dump and generated gcode attached.

I've noticed this on a number of other prints as well, but it really jumped out at me on this today as I was seeing a lot of it printing the lower layers.   Seems like a lot of those moves could be eliminated since they don't do anything and just waste time.  Unless there's some reason for them that escapes me.

bubnikv, 2017-04-24:
It may be the gap fill followed (or preceded, my memory does not serve me)
by the infill. First all the gaps are filled over a slice, second the
regions are filled in. It is true that the gap fill shall be ideally
interwoven with the infill. The only good reason for the current behavior
may be, that the gap fill changes the extrusion rate rapidly, which may
negatively influence infill lines when used with bowden extruders.

bubnikv, 2017-04-24:
Would you please provide a public youtube link to your video? I cannot access the link you provided.

jhitesma, 2017-04-24:
No problem, I had it as unlisted but changed it to public.

bubnikv, 2017-04-24:
> No problem, I had it as unlisted but changed it to public.

I think you need to post a new URL.

jhitesma, 2017-04-24:
Sorry, my mistake.  I had actually posted the wrong URL last night and didn't notice.  Here's the correct one:

https://www.youtube.com/watch?v=dl_tNpNebmo

bubnikv, 2017-05-05:
It really looks like the gap fill being processed after the infill, so the print head walks around the islands twice.

I will look into it, but the current behavior may have its logic at least for some machines. Namely, the gap fill changes the extrusion rate rapidly as each of the gap fills may have a different width. The rapid changes in extrusion rate trigger a rapid change in the nozzle, which may not be physically possible for most of the FDM printers. If the gap fill is intermixed with other extrusions on an island, this could cause underfills / overfills in the infills.

closed 2026-09-29 by prusaslicer-bot[bot]