prusa3d/PrusaSlicer #272: When Dragging objects around they dissapear

jmtome, 2017-04-24:
### Version

1.34.0 and 1.34.1 

### macOS sierra 10.12.4

### Behavior
* _Describe the problem_

When dragging objects around to arrange them, they dissapear and the platter control is lost.

* _Steps needed to reproduce the problem_

Simply import an object and begin dragging it around the platter, it will dissapear

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

no

_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[image]

[bug.ini.zip](https://github.com/prusa3d/Slic3r/files/950446/bug.ini.zip)

bubnikv, 2017-04-24:
The camera manipulation needs to be improved.

bubnikv, 2017-04-24:
This may be a similar issue to https://github.com/prusa3d/Slic3r/issues/260

bubnikv, 2017-06-08:
Would you please test the build attached to https://github.com/prusa3d/Slic3r/issues/260 ? Thanks.

bubnikv, 2017-06-14:
Closing, hopefully it is fixed in 1.35.3. If you experience this issue again, please reopen.

closed 2017-06-14 by bubnikv