prusa3d/PrusaSlicer #273: "Overhang threshold" and "Enforce support" don't work

BManx2000, 2017-04-24:
### Version
1.34.1-prusa3d

### Operating system type + version
Happens on both Windows and Mac

### Behavior
I can't generate support material for overhang angles greater than 30 degrees from horizontal. Any value I enter for "Overhang threshold" greater than 30 has no effect (but values below 30 do). "Enforce support" option also does not work for overhang greater than 30 degrees.

[image]

#### STL/Config (.ZIP) where problem occurs
[overhang.zip](https://github.com/prusa3d/Slic3r/files/952259/overhang.zip)

Hugoz12, 2017-04-24:
Seems to be an issue with layer height. If you put 0.2+ it works.

BManx2000, 2017-04-24:
I tried it some more with different layer heights, it seems that the maximum usable value for "Overhang threshold" is proportional to the layer height.

bubnikv, 2017-07-07:
Fixed with https://github.com/prusa3d/Slic3r/commit/abcd746774cbb6a392b275e331621dbb6b459ded

closed 2017-07-07 by bubnikv