prusa3d/PrusaSlicer #279: Support Generation bug for horizontal surfaces

git2212, 2017-04-26:
### Version
1.31.2 Prusa3d Win64

### Operating system type + version
Win7

### Behavior
Support generation is off for a simple case where I have 3 horizontal surfaces at varying Z offsets. 

- When overhang threshold=1, support is generated for the oval shaped surface only, in addition to undesired support structure outside the object (see oval.jpg)
- When overhang threshold=2, support is generated for the oval and hexagonal while the undesired support structure outside the object is no longer there (see oval_hex.jpg)
- When overhang threshold>3, support is generated for all three surfaces (see expected.jpg)

The expected behavior is that for overhang threshold 1 and 2, support is generated for all 3 surfaces with no undesired "artifacts" outside the object.

This appears to be a bug.

#### STL/Config (.ZIP) where problem occurs
[config.ini.TXT](https://github.com/prusa3d/Slic3r/files/958951/config.ini.TXT)

[image]
[image]
[image]

bubnikv, 2021-03-18:
Would you please provide a 3MF for this issue?

bubnikv, 2021-09-23:
No 3MF provided, closing.

closed 2021-09-23 by bubnikv