prusa3d/PrusaSlicer #281: Feature request: Layer seam removal

Quintox303, 2017-04-27:
Theres a lot of information going around about layer seam hiding, especially after Stratasys was awarded their patent on a seam hiding method (https://www.google.com/patents/US8974715).

It seems like there are certainly other solutions to the ones that they have patented, and its also only a US patent so wouldn't it be ok for Prusa to use these methods in the Czech republic? I've attached some pictures of alternative methods

[image]
 More information here http://blog.reprap.org/2009/07/build-quality.html (note that this dates back to before the patent was even filed).

I also have an idea to gradually increase the z height over a section of the perimeter such that there is no need for the print head to stop.

What are the chances of features like this being integrated in prusa slic3r?

bubnikv, 2017-05-20:
> What are the chances of features like this being integrated in prusa slic3r?

As long as it improves the prints and it is not encumbered by patents, it will likely be integrated.

Please note though, that Slic3r already does a seam optimization. It clips the loop a bit shorter and then it moves the print head slightly inside the loop after the loop is closed.

https://github.com/prusa3d/Slic3r/blob/master/xs/src/libslic3r/GCode.cpp
starting with line 1569, then at line 1600 the print head moves inside.

Quintox303, 2017-05-21:
I did some testing and I did see that! its pretty cool
[image]

I think the most interesting thing would be a gradual z height increase on the perimeters. I did some more thinking about it - turns out "layer change seam" is sort of a misconception since the layer change is usually done on the inside of the print after it prints the infill, not at the end of the outer perimeter loop (unless you're printing infill first and then perimeters, but that's not standard). 

The seam just comes from the fact that the external perimeter loop must start and end at a point - it's not completely continuous. But what if we could print the perimeters continuously? Think about printing a cylinder with 3 perimeters - but instead of the perimeters being individual concentric circles, they were joined together like an archimedean spiral (this is the closest picture I could find, imagine this but with only 3 revolutions)
[image]
The exterior revolution of the spiral would have to be modified so that the extrusion width tapered off and got thinner and thinner to create an actual circle. Ok, then you wouldn't have to start and stop the perimeter on every loop - but you would still have to stop it at the end point of the loop. And we don't want any stops on the exterior surface - we want constant motion.

So to mitigate that, what if the perimeters were printed in pairs - two layers at a time. At the end of the first layer of the pair, the outer loop of the perimeter would gradually increase in z height (like spiral vase mode) so that it would then continue onto the second layer - starting on the outside of the perimeter spiral and then moving inwards. This way the perimeter spiral would terminate on the inside of the model - leaving absolutely no seams on the outside. 

I hope that's understandable. I'll write a more detailed explanation of the concept when I have more time on my blog. Also to be clear I'm not expecting something like that to be integrated any time soon - I know it would be very complicated to develop. But if we wish to travel towards seamless (literally) printing one day then I think this could be a really good way to do it.

bubnikv, 2017-05-21:
> The exterior revolution of the spiral would have to be modified so that
the extrusion width tapered off and got thinner and thinner to create an
actual circle. Ok, then you wouldn't have to start and stop the perimeter
on every loop - but you would still have to stop it at the end point of the
loop. And we don't want any stops on the exterior surface - we want
constant motion

Also the FDM printers do not like to taper the extrusion width to zero, the
extrusion becomes unreliable, the shape of the extrusion undefined..

> So to mitigate that, what if the perimeters were printed in pairs - two
layers at a time. At the end of the first layer of the pair, the outer loop
of the perimeter would gradually increase in z height (like spiral vase
mode) so that it would then continue onto the second layer - starting on
the outside of the perimeter spiral and then moving inwards. This way the
perimeter spiral would terminate on the inside of the model - leaving
absolutely no seams on the outside.

That sounds interesting, but you will not get rid of the seam completely.
You will just decrease the number of seams by removing the seam from every
other layer.

Quintox303, 2017-05-21:
>  FDM printers do not like to taper the extrusion width to zero, the
> extrusion becomes unreliable, the shape of the extrusion undefined..

Good point about the tapering.. what if you did a combination of overlapping the perimeter that it is tapering towards, as well as lowering the extrusion width - but not all the way down to zero. I know it would result in extra filament deposition in that area but the question would be whether it would have a noticeable effect on the print quality (when you're dealing with extrusion widths below 0.5mm I think the effect would be pretty small). It would be really good if I could test this but unfortunately I have no idea how to start programming or modifying a slicer to generate a toolpath like this. I've got some programming experience but that area is a mystery to me.

> That sounds interesting, but you will not get rid of the seam completely.
> You will just decrease the number of seams by removing the seam from every
> other layer.

I acknowledge that the other suggestion of printing in pairs will not get rid of seams completely, but the idea is that the seam (on every second layer) will be positioned on the _inside_ of the perimeter spiral and not the outer surface. 

Imagine it like this.. on layer one (of the pair), the perimeter spiral is printed beginning on the inside of the spiral and moving outwards. On the outer spiral revolution, the z height is gradually increased to the next layer. From here, the spiral continues, except moving inwards for this second layer instead of outwards like the first layer was (since it's already on the outside). When the spiral perimeter is complete for the second layer, the nozzle will be on the inside of the model's perimeter loop, so the mark left here at the termination of the spiral is irrelevant because it's on the inside of the model. From here, two layers worth of infill are printed (either as one layer with double the layer height of the perimeters, or as two individual layers).  So it goes [spiral perimeter directed outwards] > [transition] > [spiral perimeter directed inwards] > [infill] > [layer change] > [repeat].

closed 2026-09-29 by prusaslicer-bot[bot]