prusa3d/PrusaSlicer #282: PRUS.cpp fails to compile in master, but is fine in stable

Jarek-SRT, 2017-04-30:
### Problem
Slic3r in branch _master_, commit 72ae3585e415189286cdb85f6e02e4ec150084ee fails to compile with the following error:
```
src/libslic3r/Format/PRUS.cpp: In function ‘bool Slic3r::load_prus(const char*, Slic3r::Model*)’:
src/libslic3r/Format/PRUS.cpp:122:40: error: conversion from ‘const char*’ to ‘const wxString’ is ambiguous
     wxFFileInputStream          in(path);
                                        ^
```
In same environment, branch _stable_, commit 2d1bf3ce439984e0f38c62560898bae30305f0a5 builds fine.

### Operating system type + version
openSUSE Leap 42.2 (x86_64) with gcc-4.8-9.61 and perl 5.18.2

### Why I think this is an issue :)
It's obvious we cannot guarantee that the code compiles in every environment, but in this case all the dependency/prerequisite checks succeed, but nevertheless compilation fails. My guess is that in master we now need a newer version of some library, so dependencies in Build.PL should be updated.

### Additional info
Output from the Build.PL checks:

App::cpanminus is up to date. (1.7043)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.10)
Encode is up to date. (2.89)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.24)
ExtUtils::ParseXS is up to date. (3.30)
skipping S/SH/SHAY/perl-5.24.1.tar.bz2
File::Spec is up to date. (3.62)
Getopt::Long is up to date. (2.49)
IO::Scalar is up to date. (2.111)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003002)
skipping S/SH/SHAY/perl-5.24.1.tar.bz2
Scalar::Util is up to date. (1.47)
Test::Harness is up to date. (3.39)
Test::More is up to date. (1.302083)
Thread::Semaphore is up to date. (2.13)
Time::HiRes is up to date. (1.9742)
Unicode::Normalize is up to date. (1.25)
threads is up to date. (2.15)
cpanm (App::cpanminus) 1.7043 on perl 5.018002 built for x86_64-linux-thread-multi
...
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Checking if you have Module::Build 0.38 ... Yes (0.4003)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have Test::More 0 ... Yes (1.302083)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have Module::Build 0.3601 ... Yes (0.4003)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)

bubnikv, 2017-06-02:
I suppose Alien::Wx module pulls the wxWidgets library of older version than expected. I think your wxWidgets library is older than 3.0.

I guess though that you could compile Slic3r with the older wxWidgets library if you just explicitely convert to wxString
> src/libslic3r/Format/PRUS.cpp: In function ‘bool Slic3r::load_prus(const char*, Slic3r::Model*)’:
< src/libslic3r/Format/PRUS.cpp:122:40: error: conversion from ‘const char*’ to ‘const wxString’ is ambiguous
     wxFFileInputStream          in(path);

What if you change the line
     wxFFileInputStream          in(path);
to
     wxFFileInputStream          in(wxString(path));
? 

Will it compile?

wrowe, 2017-06-23:
"I think your wxWidgets library is older than 3.0."

wrowe, 2017-06-23:
"I think your wxWidgets library is older than 3.0."

Upstream, wx 2.8 is supported by slic3r.

Fedora 25 (and SuSE 42.2, apparently) ship wx 2.8.

For Fedora, there is a wx 3.0 parallel distribution.

However... Fedora builds of Perl modules target 2.8.

My error log is slightly longer than OP's... but I presume the answer is "use wx 3.0" for prusa fork?
[err_log.txt](https://github.com/prusa3d/Slic3r/files/1098357/err_log.txt)

bubnikv, 2017-06-23:
> My error log is slightly longer than OP's... but I presume the answer is "use wx 3.0" for prusa fork?

Yes. We have to look into the future.

wrowe, 2017-06-23:
Sensible - but a particular shame since all the docs point users to;
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

Which of course references 2.8, because "it just works" with the older wx.

Those references should probably be annotated, particularly for users
such as I who need to begin by deinstalling all system wx-* and alien-wx
components, removing any cpan-provisioned flavors, and rebuilding from
scratch. Fedora 26 isn't even released yet, and yet Fedora Perl community
is still glued to that flavor. So looking ahead is delightful, but gets a number
of users nowhere since wx 2.8 <> 3.0 was a somewhat radically breaking
change that takes a number of years to accommodate.

wrowe, 2017-06-26:
Also in Fedora 25 after ripping out Fedora builds of wx components and wxGTK (2.8), the Alien::wxWidgets seems to build fine and detect wxGTK3-devel, but then Wx 0.9932 fails to build against Fedora's wxGTK3 distro for the PropertyGrid.cpp source. This might be a local EBCAD issue since I haven't seen reference to such an incompatibility yet, but just a heads-up that Fedora users' mileage may vary installing these themselves from cpan.

bubnikv, 2017-07-07:
@wrowe Unfortunately there are roughly 150 open issues, which I need to handle. I would be most happy, if you could help out to rectify this ugly mess and provide a pull request for the build documentation.

Regarding Linux installations, every Linux user has his own distro. It is not in a power of a single man to develop an application and to support every possible Linux distro. Therefore we prepared an appimage.

closed 2017-10-03 by bubnikv