prusa3d/PrusaSlicer #283: Request: assign separate key combinations to "Re-slice now" and "1-layer"

szalat, 2017-05-02:
### Version
1.33.8_devel
# git describe --tag
0.17_pre-4258-gdf1e2642d

### Operating system type + version
Linux 4.9.0-2-amd64 #1 SMP Debian 4.9.13-1 (2017-02-27) x86_64 GNU/Linux

### Behavior
The issue is simple to fix. It is about one shortcut connected to different actions according to opened tab. 
In 3D tab shortcut CTRL-S issue an "Re-slice now"  command, but same key combination pressed in "Preview" tab switch on or off "1 layer" checkbox.
"Re-slice now" it is huge improvement although It is annoying when I try to adjust some parameters, reslice model and check results in preview - each time I must go to "3D" tab, press CTRL-S and then just switch to "Preview".

I think one of this two operations needs different keyboard shortcut.

bubnikv, 2017-06-06:
The two key combinations are different: Ctrl-S for slice now and S for "single" layer.
It works perfectly on Windows, but it is somehow broken on Linux.

bubnikv, 2017-06-06:
Would you please test with the Slic3r PE 1.35.2 release? I have changed the keyboard handling a bit: I am testing explicitly for the modifier keys (ctrl, alt) and I am passing the key press messages to the rest of the application if not processed by the registered handlers.

bubnikv, 2017-10-03:
no response. I hope this has been fixed already.

closed 2017-10-03 by bubnikv