prusa3d/PrusaSlicer #284: [Request]Turn on fan on Overhang Threshold

sarf2k4, 2017-05-02:
I would like to request this feature since I noticed using fan for overhang helps a lot in retaining its shape without curling.

I personally print with abs but turning fan on for lower overhang like lesser than 35 degree, able to prevent curling. Since abs aren't good to be paired with fan, adding this feature will enhance print quality for lower overhangs.

For this feature, it should be separate with support but almost similar to it. It will be independent and has its own overhang threshold, fan speed range and etc (I can't think of any yet)

Sebastianv650, 2017-05-02:
I described this request here #242. I'm optimistic bubnikv will add something like that one day :)

bubnikv, 2017-07-07:
Closing this issue as it is a duplicate of #242.

closed 2017-07-07 by bubnikv