prusa3d/PrusaSlicer #285: "Arrange" does not arrange model that appears to fit platform

fiveangle, 2017-05-02:
### Version
_Version of Slic3r used goes here_

1.34.1 2017apr06-0802

### Operating system type + version

Mac OS 10.12.4

### Behavior

"Arrange" does not arrange model that appears to fit platform

* _Steps needed to reproduce the problem_

1. Import [SAC_Wrench.stl.zip](https://github.com/prusa3d/Slic3r/files/970955/SAC_Wrench.stl.zip)
2. Click Rotate 45 deg CCW button
3. Click Scale button and enter 125%
4. Click Arrange button

* _Expected Results_

Model centered on build platform

* _Actual Results_

Model unmoved

[image]

_Is this a new feature request?_

No

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/970960/config.ini.zip)

bubnikv, 2017-05-08:
I confirm your findings.

It seems to be a problem of the way Slic3r handles the rotation / scaling internally. If you you export the rotated and scaled STL, then import and click arrange, it will work as expected.

beast-CAT, 2017-05-26:
I having the same issue with the spit turn table parts 2 and 3 from the $30 3D Scanner (https://www.thingiverse.com/thing:1762299) after laying it flat and rotating it 45°.
[image]

tkircher, 2017-05-28:
I'm also seeing this issue, in 1.34.1.24-prusa3d. This is a regression, since it mostly works in 1.33.8-prusa3d, but seems to do literally nothing in the new version.

bardiir, 2017-06-09:
I'd propose a different fix for this - make arrange non-mandatory for single parts. Sometimes you want to print something not exactly in the center of the printbed, for example when I test new filaments I'm printing to the side of the printbed as i can more easily get it off at the edge, but Slic3r wont let me put an object at the edge, only the center. Manually rearranging an object would help to work around this issue too as you could manually fix the misalignment.

tkircher, 2017-06-09:
I checked 1.33.8-prusa3d and 1.35.2.4-prusa3d, and I don't have any problem moving single parts around arbitrarily on the print bed.

Hugoz12, 2017-06-11:
You have to disable "Auto-center parts" in "File" -> "Preferences".

bardiir, 2017-06-11:
Oh... never noticed that, thanks :rofl:

bubnikv, 2017-06-13:
Fixed with https://github.com/prusa3d/Slic3r/commit/5cae4cc614dc379b35bba6397ea2bf83f0adac56
It was an old issue, not a regression.

closed 2017-06-13 by bubnikv