prusa3d/PrusaSlicer #286: Add Brim to object settings

jindrichbenes, 2017-05-03:
Add Brim to object settings.

[image]

[image]

jltx1, 2017-05-11:
I've had the same request in the main slic3r branch.  Hopefully one of them gets it in.

bubnikv, 2017-06-02:
This is not so easy as Slic3r currently calculates brim & skirt for all objects at once.

jltx1, 2017-06-02:
makes sense for skirt.  For brim, can you consider the footprint of the non-brim objects as already having completed their brim expansion?  I notice there is brim collision handling already.

bubnikv, 2017-06-14:
Also there is a request for the multi-material to print the brim with the same extruder as the object, so the colors will not be mixed.

jhoblitt, 2019-02-06:
I would very much appreciate being able to set a brim on a per object basis.  Currently, if I'm building a parts kit that has at least one "risky" part, I need split it into two builds: easy parts and a parts that need brims.  Otherwise, it is a choice between brims everywhere or risking a tall/thin part going "wrong" and ruining the entire print.

jindrichbenes, 2019-02-07:
Slic3r core does not support brim for selected parts, what works is using add part > cylinder (i.e.) > set Z height to 0.2, set other dimensions as seemed fit > move to bed. You now have effectively a brim.

mdealer, 2019-03-10:
I also need this as my printer tackles many different objects at once on a large bed. The ability to change stuff per-object is really welcome.

fe60, 2020-05-29:
@mgx0 This request is about brim settings per model [1], not brim in general.

Even though this issue older than #1033 I still propose to close this one as duplicate and continue in #1033. Its initial entry got more than 20 thumbs up already.

[1] https://help.prusa3d.com/en/article/per-model-settings_1674

birtalanelod, 2021-04-19:
modifier would be great, when printing multiple objects only add to the tall ones

RichiH, 2021-05-15:
My current thinking is to simply ask about this feature request during the next Prusa Live; this worked quite well for https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/739

jltx1, 2021-05-15:
I use a simple work around that is just as effective. In PrusaSlicer, right click on plater and add shape, like cylinder.  Use cut and type in 0.2 (first layer height) and keep lower.  Now you can slide this puck under any tall object or thin parts of other objects.  You can hit ‘+’ at top to make many copies and scale X and Y as needed.  Honestly, in many ways it is better than brim because you put it only where needed so less to clean up post print.

lukasmatena, 2021-08-12:
Upcoming PrusaSlicer 2.4.0-alpha1 will allow to set brim for each object separately. Closing.

closed 2021-08-12 by lukasmatena

lukasmatena, 2021-10-21:
@thwerks You somehow forgot to enclose a 3MF, so there is no way to tell if you set it correctly, or just assumed slicer will magically guess which object you want the brim on. I just tested in 2.4.0-alpha1 and 2.3.0-alpha3 and don't see any issue.
[image]