prusa3d/PrusaSlicer #288: nozzle crash in the print during the print

marc2085, 2017-05-06:
### Version
1.34.1
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
mac osx

### Behavior
* _Describe the problem_ the nozzle crash in the print at 15.5mm

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-05-08:
You provided too little information for me to reproduce. Would you please provide your config file and the STL, maybe a G-code as well?

marc2085, 2017-05-08:
sorry , i have upload my configuration with stl, g code , config and picture of the part

bubnikv, 2017-06-02:
Thanks, I gave your g-code to our print expert guy to run it. Let's see.

bubnikv, 2017-06-06:
We printed the same piece and we did not have any issue with the G-code you have provided.

bubnikv, 2017-06-08:
It may actually be a hardware problem. I have seen situations, where the Z axis had not enough strength to lift up in some cases, but it moved down happily. If a Z-up movement came after a rapid movement in the XY plane, the Z axis was jammed by the jerk in the XY axis and it did not move up far enough. Then the successive z-hop down movement sank the nozzle into the print.

marc2085, 2017-06-09:
I have upgrade the firmware with the lastest version  last week . I will see if i have any crash in future . Thanks for the information

closed 2017-06-09 by marc2085