prusa3d/PrusaSlicer #289: Error on startup due to Mklink

m-mccormick, 2017-05-07:
### Version
1.34.1-prusa3d-win64

### Operating system type + version
Win 7 SP1 x64

### Behavior
I have a mklink set up pointing "C:\Users\User\AppData\Roaming\Slic3r", "Slic3r" being the mklink, to a directory on a different disk "S:\Files\Slic3r", so that the same files can be used on multiple OSs on a multiboot machine. When starting slic3r, I get a dialog box stating "Error: Slic3r was unable to create its data directory at C:\Users\User\AppData\Roaming\Slic3r (File exists)." This is likely because directory and file exists functions usually don't take mklinks into consideration.

Not a new feature request

#### STL/Config (.ZIP) where problem occurs
Not applicable

BirkBinnard, 2017-05-08:
I am not sure what an mkink is, so this is just a guess, but I have used junction links to do similar things with no problems. There is a nice utility called Junction Link Magic that does a nice job of listing, creating, and deleting junction links. So you might want to try that and see if it fixes your problem.

bubnikv, 2017-05-08:
I suppose you used 
mklink /d
to create a symbolic directory link.

Will a hard link
mklink /h

work for you?

Sorry, but this is a very marginal issue and I have much more important stuff to do. The symbolic links are pretty messy on Windows.

m-mccormick, 2017-05-08:
@bubnikv Correct on mklink /d. mklink /h will not work for me due to spanning partitions. I understand it being a marginal issue and how messy it is, and where it likely falls in terms of priority. No worries, and thank you for the direct answer.

I know slic3r is written in perl, and C# likely isn't very helpful, but here is a functional snippet if it helps anyone else:
`

        public static bool GetSymbolicLink(string link, out SymbolicLink symbolicLink)
        {
            symbolicLink = null;

            // Check if file exists
            if (!File.Exists(link) && !Directory.Exists(link))
                return false;

            // Get the symbolic link type
            FileInfo fileInfo = new FileInfo(link);
            SymbolicLinkType type = SymbolicLinkType.File;
            if (fileInfo.Attributes.HasFlag(FileAttributes.Directory))
                type = SymbolicLinkType.Directory;

            string path = string.Empty;
            string name = string.Empty;
            string target = null;

            if (type == SymbolicLinkType.File)
            {
                FileInfo targetFileInfo = new FileInfo(link);
                path = targetFileInfo.DirectoryName;
                name = Path.GetFileName(link);
            }
            else
            {
                DirectoryInfo directoryInfo = new DirectoryInfo(link);
                if (directoryInfo.Parent == null)
                    return false;
                path = directoryInfo.Parent.FullName;
                name = Path.GetFileName(link);
            }

            // Get Info from Dir Command in Command Prompt
            ProcessStartInfo processStartInfo = new ProcessStartInfo
                                                {
                                                    CreateNoWindow = true,
                                                    UseShellExecute = false,
                                                    FileName = "cmd.exe",
                                                    WindowStyle = ProcessWindowStyle.Hidden,
                                                    Arguments = "/c dir " + "\"" + path + "\"",
                                                    RedirectStandardOutput = true,
                                                    //RedirectStandardError = true
                                                };
            Process process = Process.Start(processStartInfo);
            List<string> symLinks = new List<string>();
            using (StreamReader streamReader = process.StandardOutput)
            {
                string output = streamReader.ReadToEnd();
                string[] lines = output.Split('\n');
                foreach (string line in lines)
                {
                    if (line.Contains("SYMLINK"))
                    {
                        if (line.Contains(" " + name + " "))
                        {
                            // Parse the string to get the
                            int startPoint = line.IndexOf(name) + name.Length;
                            target = line.Substring(startPoint, line.Length - startPoint);
                            target = target.Replace("[", "").Replace("]", "").Trim();
                        }
                       
                    }
                }
            }

            if (target != null)
            {
                symbolicLink = new SymbolicLink(link, target);
                return true;
            }
            return false;
        }

`

closed 2017-05-08 by m-mccormick

bubnikv, 2017-05-12:
> I know slic3r is written in perl, and C# likely isn't very helpful, but
here is a functional snippet if it helps anyone else:

That's why the link is not used on Windows, because it is not transparent
to the applications.