prusa3d/PrusaSlicer #290: variable fill density

mischlrebl, 2017-05-08:
I am quite sure, that this has been discussed already somewhere but I as a newbie am not able to find it... so perhaps it is a new feature request:
according to the famous **Variable Layer Height** a similar **Variable Fill Density** would be awesome.

like the layer height the infill density - for simplicity only the percentage - should be adjustable - only increasing the density - the exactly same way. there are a lot of models with different requirements to the infill at different regions. because it is difficult to separate individual parts of an object it should be unified to layers again.

I have read about it once: https://www.simplify3d.com/support/articles/different-settings-for-different-regions-of-a-model/
but the use of the variable layer height is much more intuitive and a lot faster. 

In my simple imagination, the fill density and its pattern should be like defined in the print settings and as it would made the way now (for example the cubic infill). if on some layers a higher density is wished, the grid will smoothly extended before (and perhaps after) with 45° structures in additional directions of xy to hold and build on the additional sub grid. like mentioned for the surface under http://www.prusaprinters.org/introducing-slic3r-prusa-edition/ where it is stated "Slic3r now makes sure it lays down the needed supports for perimeters in upcoming layer".

probably the approach with percentage is not correct and it should be more like: dividing the pattern by n times.
if the grid would be finer on a specific height this would not strengthen the part if in the middle (in z) of a block but when set on some bulges until the the end (outwall), I guess.

[image]

jhitesma, 2017-05-08:
slic3r already supports variable fill density by using modifier meshes:  http://slic3r.org/blog/modifier-meshes

It's not super obvious since it requires multiple .stl's but it's more powerful than the method s3d uses based on my reading of their offering.  I've used it a couple times to selectively strengthen sections of a print that I wanted to keep lightweight but needed stronger sections on.

mischlrebl, 2017-05-09:
thank you for pointing this - I would never have been able to find it.

but it does not work... or I am not able to use it what is much more likely

I set the infill density to 10% in the Print Settings (trap: with the default pattern stars a 100% infill is not possible. the error message appears only in the Print Settings tab, not in the Settings popup. used rectiliniear)
[image]

placed the modifier box (trap: it is easily that not only the modifier is selected but the whole object...)
[image]
the placing is only possible with the three sliders here or? no drag and drop anywhere?

reslice again
[image]
the whole infill is now affected with 100%, why?

this leads to the question if only one of the two infill volumes will be affected, what will happen then. that work's:
[image]

conclusion: this works 'only' on an infill which is completely outlined by perimeters. correct?
and how to reach this in a design to force a separate section for this? in my opinion the modifier meshes should be placed anywhere without regards to infill and perimeters...

jhitesma, 2017-05-12:
It does seem that something has changed here.  I just tested the model I've used this one before and it's doing the same thing.  I'm getting more infill outside of the modifier as well as within it while before the additional infill was limited to within the modifier.  

This piece:
https://github.com/jhitesma/jth_angle_quad/blob/dtfc_mods/generated_files/angle_quad_dtfc.stl

With this modifier:
https://github.com/jhitesma/jth_angle_quad/blob/dtfc_mods/generated_files/center_infill.stl

Here's gcode generated with 1.33.5-prusa3d back in Feb:
[angle_quad_dtfc.gcode.zip](https://github.com/prusa3d/Slic3r/files/997509/angle_quad_dtfc.gcode.zip)

Looking at it with gcode.ws I can see that only the center is at 100% infill:
[image]

The same files sliced with 1.34.0 today I'm getting 100% infill everywhere:
[image]

So something seems to have changed since 1.33.5 and 1.34.0 that broke modifier meshes

jhitesma, 2017-05-12:
Just noticed I didn't use the same settings on that test as the original had 4 perimeters and the new test only had 3.  Redid it with the old settings and got the same results.  Just to rule that out.

bubnikv, 2017-06-06:
@jhitesma would you please provide the AMF file and the config.ini? This way I can repeat your scenario exactly. The AMF file contains all the multi volumes, modifier volumes with their placement and the override parameters of the modifier volumes. Thanks.

jhitesma, 2017-06-06:
@bubnikv be glad to.  Where can I find the AMF file?  (on macos)  I've never heard a reference to that before and just used the GUI to setup the modifier volume.

bubnikv, 2017-06-06:
Menu platter-> export as AMF.
The AMF import / export is quite usable now since it has been rewritten to C++.

jhitesma, 2017-06-06:
Just tried again (with 1.34.1.24) and...it worked as expected :)

[image]

So either I was doing something wrong repeatedly last month or something between 1.34.1 and 1.34.1.24 fixed it.  Either way I can no longer reproduce 👍 

@mischlrebl can you still reproduce with the files you were using?

fiveangle, 2017-06-07:
Looks like a regression was resovled, which is great.  But I still agree with the premise of your request, and would go so far as to suggest that at some point in the future, an option to allow detection of sharp interior angles or thinner sections, and automatically increase infill percentage in those higher stress areas (something like a "Strength Mode Infill" would be a leap forward for reducing print time, material use, and weight, while keeping strength in the most obvious critical areas.  Or more importantly, reducing the infill percentange for large volumous areas would probably be an easier first target for such a feature.

closed 2026-09-29 by prusaslicer-bot[bot]