prusa3d/PrusaSlicer #291: Ctrl+1 misbehaviour

mischlrebl, 2017-05-09:
### Version
Slic3r Prusa Edition
Version: 1.34.1-prusa3d-win64
Build: Slic3r-1.34.1-prusa3d-win64-full-201704060808
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Operating system type + version
Win 7 Pro, SP 1, 64 Bit OS

Location: Switzerland, German language

### Behavior
Ctrl+1 works? in the Print Settings as in the other Tabs nothing happens but the copies on the Plater are increased (Ctrl++). In my region, the + sign is the shift position on the 1 key.
Ctrl+(2-4) works well

bubnikv, 2017-06-06:
I understand your problem, but I am afraid I cannot do much as of now, as the bug seems to be inside the wxWidgets user interface library. At least I don't see a simple bugfix.

bubnikv, 2017-07-25:
Would you please test with the just relased Slic3r 1.36.1?
I have changed the Ctrl++ shortcut for object duplication to just '+', so your problem should be gone.
Thanks.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.36.1

bubnikv, 2017-10-03:
No response. I believe it has been fixed.

closed 2017-10-03 by bubnikv