prusa3d/PrusaSlicer #292: sequential printing not working

jansmets, 2017-05-09:
### Version
1.34.1

### Operating system type + version
Win 7 

### Behavior
I did enable sequential printing, loaded up stl files, creating multiple groups.
Then when sliced it seems that the layers of all parts are printed at the same time. 
I also loaded them up in octoprint and verified the same.

bubnikv, 2017-05-20:
Do I understand it correctly, that the sequential printing works for the objects, but not for their brim & skirt?

jansmets, 2017-05-20:
Yes. Correct

michalzielanski, 2017-05-29:
@jansmets How to create a `gcode` file with sequential print?

I select `Complete individual objects` in `Printing Settings > Output options > Sequential printing`; I add two `stl` files to the `Plater`, place objects far apart, create a `gcode` file (`Export G-code...`) and when I load created `gcode` file to the [gcode.ws](http://gcode.ws), I see that the print goes parallel rather than sequentially.

bubnikv, 2017-06-06:
This is a similar issue to
https://github.com/prusa3d/Slic3r/issues/286

AmiableAnteater, 2017-06-09:
@michalzielanski gcode.ws (and other tools) apparently show the gcode by layer. I was confused by this as well. But if you actually print objects with the option `Complete individual objects` they are printed sequentially.

+1 for this enhancement request.

dewi-ny-je, 2021-02-25:
The feature keep popping up in the forums.
https://forum.prusaprinters.org/forum/prusaslicer/complete-individual-objects-should-include-the-brim

It seems like a poor design choice which should be corrected.

keeena, 2021-03-15:
Would love to see this feature addressed (brim & skirt printing sequentially when "Complete Individual Objects" is selected).

bbrendon, 2021-06-11:
For anyone looking for a work-around. I just loaded my prusaslicer file into SuperSlicer and it doesn't have this issue. Worked as expected!

thwerks, 2021-10-22:
> For anyone looking for a work-around. I just loaded my prusaslicer file into SuperSlicer and it doesn't have this issue. Worked as expected!

Thank you for mentioning SuperSlicer, very interesting slicer but for me its not possible to use. When I open my PrusaSlicer projects in SuperSlicer I get a completely different result and not for the better. I find that no matter what settings I put in SuperSlicer I cant get acceptable results. Cool project but at this time its flaws are too numerous to count and its almost useless in my opinion. 

Anyway, after 4 years of no improvement from PrusaSlicer in this regard, I have made my own GCode modifier program that fixes this issue as well as some additional features for multi-part printing. 

You can check it out and download it from here: https://github.com/prusa3d/PrusaSlicer/issues/7145

thwerks, 2022-06-15:
It is baffling how PrusaSlicer handles brims and skirts; despite YEARS of dozens and dozens of people repeatedly complaining about this blatant flaw.  I don't even bother submitting bugs and recommendations here because its just a waste of time. The PrusaSlicer opensource project is Prusaheaded (hardheaded and egotistical). No surprise, what kind of person names every single product after himself, that's just cringe.

Any way, i have a utility that fixes this flaw in some cases, in that a single part with a skirt or brim is multiplied. Moreover it automatically arranges the parts on the bed the the maximum that will safely fit. It wont help for dissimilar sequentially printed parts at the moment by if there is interest i might do that. 

you can learn more here:
https://github.com/prusa3d/PrusaSlicer/issues/7145

jhenley01, 2022-10-04:
Running version 2.5 and this is still an issue, SuperSlicer handles this way better but like others have said importing projects from PrusaSlicer to SuperSlicer often leads to different results when printing.

For me the main reason you want the skirt or brim to print with each object is because of nozzle ooze. By the time the  print head moves over and down to the bed to start the next part you can have several mm of ooze leak out ot the nozzle, especially with PETG. Much better to have that ooze in your skirt or brim vs your part. 

For sequential printing It would also be nice to be able to rename or order the instances so you can print them in the order you want vs having to import multiple versions of the same model. Sequential printing can be very useful but seems like it's often ignored.

konopik, 2022-10-18:
+1 Still a problem in 2.5

cimbalek, 2022-12-04:
+1 Upvote.
I'd like to use brim for priming the nozzle when priting multiple objects with heavily oozing material (LW PLA). I need to separate and print per object or bear incomplete first layer.

Tupson444, 2023-02-06:
When printing sequentially, I will (now that I realised it) actually use brims to be printed all at once in the first layer, in order to make sure each part is sticking well to the bed (I have manual levelling, so it would be a final check whether the bed is level). If all the brims look good, then I'm confident that the whole print will go well.

However, I consider the above example to be 'alternative' use, while the primary use should be as expected: if sequential printing is turned on, the brims should print sequentially as well.

It would be good to be able to choose between the two options, like in SuperSlicer where you have "Allow only one skirt loop" and "Print all brim at startup" checkboxes.

iamr1nce, 2023-02-13:
This google search journey was a wild one. So many requests and all roads lead back to this thread.

Adding another voice to the chorus. Seems like the best place to place these check-boxes would be : Print Settings > Output options > Sequential printing > "Complete individual skirts" & "Complete individual brims"

Loving 2.6.0-alpha3 so far, and would definitely appreciate having this slide in to the beta and subsequent full release as an unexpected bonus. =)

thwerks, 2023-02-13:
Even if PrusaSlicer eventually resolves this issue, someone may still benefit from the auto arranging capability of this ustility.

https://github.com/thwerks/GCode-Auto-Arranger

Sp4rkR4t, 2023-03-26:
Still an issue in Prusaslicer 2.6-alpha 5

AdeAnima, 2023-06-20:
Wow only an issue since 6 years ^^

alexpantone, 2024-04-10:
I am still having this issue on 2.7.4.

Snuff1eupagus, 2024-04-28:
Try this,  I'm pretty sure it will help

https://www.youtube.com/watch?v=dhsbMmosbLU&t=1s

I can't believe this guy waited 3 years to show us this. What did we do wrong?

stefanct, 2024-06-14:
Has anybody looked at the source code yet? This seems like a relatively simple thing to do on a high level. There are no customizations to be taken into account (apart from the literal custom g-code between objects) so it's really just a matter of splitting the slicing of the brim layer into individual chunks that get emitted at the right place, i.e., before the respective object's code. Now, in reality the slicer is convoluted no question but this is github after all. If my todo list wouldnt be as long I'd give it a shot :)

Cinderhaze, 2024-11-07:
Came across this issue - I have the MMU, and I want to do something relatively simple to create filament swatches for myself - load up 5 filaments, put them all on the bed, print sequentially with different extruders with some kind of purge between each print... it's amazing that this is still something that can't easily be done.

langusta, 2025-03-19:
This is kind of cruicial for multiple lw pla prints. It's a pity that it's been ignored over so many years..

dewi-ny-je, 2025-11-12:
> The issue persists in version 2.9.3; a TPU gasket (without brim) prints first, followed by two PLA pieces. However, the brim for the PLA pieces is printed out of TPU and at the same time as the gasket, rather than true sequential printing.

Can you try if Orca Slicer does it differently? I think last year something was changed there

zandr, 2026-09-02:
This issue still applies to 3.0.0a11.