prusa3d/PrusaSlicer #293: After adjusting layers only the support is left

3axis, 2017-05-09:
### Version
_Version of Slic3r used goes here_

Version 1.34.1-prusa3d-linux64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

Ubuntu with Gnome shell, 17.04. System info in prusa slic3r would not copy to clipboard.

### Behavior
* _Describe the problem_

Having added stl objects and arranged them, I set the print settings to 0.2 first layer then 0.3 as I have before, I then modified the layers to make some fully fine. When I return to the preview or layers, the objects are gone and only the skirt and support remains. I had just discovered this whilst printing. The objects still show in the 3D view but not the preview or layers.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[image]
[image]

[Temp.zip](https://github.com/prusa3d/Slic3r/files/987512/Temp.zip)

bubnikv, 2017-06-06:
I believe this bug is specific to the "Background processing" feature, which we do not recommend to use, as it makes Slic3r unstable. Anyway, I will look into it with one day.

bubnikv, 2017-06-12:
This is hopefully fixed with https://github.com/prusa3d/Slic3r/commit/958c6553e79737895c8e63d66350d118b722be09

closed 2017-06-12 by bubnikv