prusa3d/PrusaSlicer #294: Crash when sending to Octoprint redirecting from HTTP to HTTPS

yschroeder, 2017-05-10:
### Version
1.34.1

### Operating system type + version
Arch Linux Kernel 4.10.13

### Behavior
* Slic3r crashes when pressing `Send to printer` with enabled redirect from HTTP to HTTPS in haproxy (octopi)

* _Reproduction_
  * Activate redirection to HTTPS from HTTP in haproxy (octopi) via adding
    ```
    redirect scheme https code 301 if !{ ssl_fc }
    ```
    to `/etc/haproxy/haproxy.cfg`.
  * Press `Send to printer`
  * Slic3r crashes without sending the gcode to the printer
* _Expected Results_
  * Slic3r does not crash but sends gcode via HTTPS to the printer
* _Actual Results_
  * Slic3r crashes (application closes without error message)

#### STL/Config (.ZIP) where problem occurs
Problem is independent from STL file used.

bubnikv, 2017-05-12:
Did you compile slic3r yourself, or do you run the appimage?
Would you please start Slic3r from the command line, let it crash and post
the command line output here?
Thanks.

yschroeder, 2017-05-15:
I am running the appimage.

Here is the output of the crash:
```
Can't locate URI/_foreign.pm in @INC (you may need to install the URI::_foreign module) (@INC contains: /tmp/.mount_p58HQ2/usr/bin/bin/lib /tmp/.mount_p58HQ2/usr/bin/lib/site_perl/5.22.0/x86_64-linux-thread-multi /tmp/.mount_p58HQ2/usr/bin/lib/site_perl/5.22.0 /tmp/.mount_p58HQ2/usr/bin/lib/5.22.0/x86_64-linux-thread-multi /tmp/.mount_p58HQ2/usr/bin/lib/5.22.0 .) at /tmp/.mount_p58HQ2/usr/bin/lib/site_perl/5.22.0/URI.pm line 61.
```

bubnikv, 2017-05-15:
Thanks, I have to get octoprint running with https redirect somewhere to
test on.

bubnikv, 2017-06-06:
This may be related: https://github.com/alexrj/Slic3r/issues/3203

bubnikv, 2017-06-14:
I am afraid I cannot invest time into it right now, as the integration of SSL on our Windows build would require likely couple of days.

I will update our binary packages though for the Slic3r to not crash, but a following window will be displayed:

[image]

vojtechkral, 2018-03-15:
There is a new implementation of Slic3r's http communication via libcurl, comming up in the following release. It should be able to handle redirects, https or otherwise. If it doesn't work in the future, feel free to open a new issue.

closed 2018-03-15 by vojtechkral