prusa3d/PrusaSlicer #295: TBB was not find.

dobrzann, 2017-05-10:
### Version
b92aa20 

### Operating system type + version
Windows 10 x64

### Behavior
I was trying to compile slic3r on two diferent pc's, diferent prel versions alsow using VSC 2013 and in all this cases Build.pl script can't find my extracted TBB.
Even adding TBB variables (TBB_DIR, TBB_INCLUDEDIR, TBB_LIBRARYDIR) is no helping. 

[image]

I was trying this TBB: [https://bintray.com/bubnikv/wxPerl-for-Slic3r-with-VisualStudio/download_file?file_path=tbb-2017_U5.7z]()

And this: [https://github.com/01org/tbb/releases]()

_Is this a new feature request?_
NO

bubnikv, 2017-05-11:
> VisualStudio/download_file?file_path=tbb-2017_U5.7z

I have the TBB environment variables set to:

set TBB_DIR=C:\local\tbb
set TBB_INCLUDEDIR=%TBB_DIR%\include
set TBB_LIBRARYDIR=%TBB_DIR%\lib64-msvc-12.0\Debug

for the debug build and

set TBB_LIBRARYDIR=%TBB_DIR%\lib64-msvc-12.0\Release

for the release build,

and the content of the folder
C:\local\tbb\lib64-msvc-12.0\Debug

is the following:

tbbmalloc_debug.lib
tbb_debug.lib
tbb_static_debug.lib

C:\local\tbb\lib64-msvc-12.0\Release

is the following:

tbbmalloc_proxy_static.lib
tbbmalloc_static.lib
tbb_static.lib

and the content of
C:\local\tbb\include

is the following:

C:\local\tbb\include\index.html
C:\local\tbb\include\out.tt
C:\local\tbb\include\serial
C:\local\tbb\include\tbb
C:\local\tbb\include\serial\tbb
C:\local\tbb\include\serial\tbb\parallel_for.h
C:\local\tbb\include\serial\tbb\tbb_annotate.h
C:\local\tbb\include\tbb\aggregator.h
C:\local\tbb\include\tbb\aligned_space.h
C:\local\tbb\include\tbb\atomic.h
C:\local\tbb\include\tbb\blocked_range.h
C:\local\tbb\include\tbb\blocked_range2d.h
C:\local\tbb\include\tbb\blocked_range3d.h
C:\local\tbb\include\tbb\cache_aligned_allocator.h
C:\local\tbb\include\tbb\combinable.h
C:\local\tbb\include\tbb\compat
C:\local\tbb\include\tbb\concurrent_hash_map.h
C:\local\tbb\include\tbb\concurrent_lru_cache.h
C:\local\tbb\include\tbb\concurrent_priority_queue.h
C:\local\tbb\include\tbb\concurrent_queue.h
C:\local\tbb\include\tbb\concurrent_unordered_map.h
C:\local\tbb\include\tbb\concurrent_unordered_set.h
C:\local\tbb\include\tbb\concurrent_vector.h
C:\local\tbb\include\tbb\critical_section.h
C:\local\tbb\include\tbb\enumerable_thread_specific.h
C:\local\tbb\include\tbb\flow_graph.h
C:\local\tbb\include\tbb\flow_graph_abstractions.h
C:\local\tbb\include\tbb\flow_graph_opencl_node.h
C:\local\tbb\include\tbb\gfx_factory.h
C:\local\tbb\include\tbb\global_control.h
C:\local\tbb\include\tbb\index.html
C:\local\tbb\include\tbb\internal
C:\local\tbb\include\tbb\machine
C:\local\tbb\include\tbb\memory_pool.h
C:\local\tbb\include\tbb\mutex.h
C:\local\tbb\include\tbb\null_mutex.h
C:\local\tbb\include\tbb\null_rw_mutex.h
C:\local\tbb\include\tbb\parallel_do.h
C:\local\tbb\include\tbb\parallel_for.h
C:\local\tbb\include\tbb\parallel_for_each.h
C:\local\tbb\include\tbb\parallel_invoke.h
C:\local\tbb\include\tbb\parallel_reduce.h
C:\local\tbb\include\tbb\parallel_scan.h
C:\local\tbb\include\tbb\parallel_sort.h
C:\local\tbb\include\tbb\parallel_while.h
C:\local\tbb\include\tbb\partitioner.h
C:\local\tbb\include\tbb\pipeline.h
C:\local\tbb\include\tbb\queuing_mutex.h
C:\local\tbb\include\tbb\queuing_rw_mutex.h
C:\local\tbb\include\tbb\reader_writer_lock.h
C:\local\tbb\include\tbb\recursive_mutex.h
C:\local\tbb\include\tbb\runtime_loader.h
C:\local\tbb\include\tbb\scalable_allocator.h
C:\local\tbb\include\tbb\spin_mutex.h
C:\local\tbb\include\tbb\spin_rw_mutex.h
C:\local\tbb\include\tbb\task.h
C:\local\tbb\include\tbb\task_arena.h
C:\local\tbb\include\tbb\task_group.h
C:\local\tbb\include\tbb\task_scheduler_init.h
C:\local\tbb\include\tbb\task_scheduler_observer.h
C:\local\tbb\include\tbb\tbb.h
C:\local\tbb\include\tbb\tbbmalloc_proxy.h
C:\local\tbb\include\tbb\tbb_allocator.h
C:\local\tbb\include\tbb\tbb_config.h
C:\local\tbb\include\tbb\tbb_exception.h
C:\local\tbb\include\tbb\tbb_machine.h
C:\local\tbb\include\tbb\tbb_profiling.h
C:\local\tbb\include\tbb\tbb_stddef.h
C:\local\tbb\include\tbb\tbb_thread.h
C:\local\tbb\include\tbb\tick_count.h
C:\local\tbb\include\tbb\compat\condition_variable
C:\local\tbb\include\tbb\compat\ppl.h
C:\local\tbb\include\tbb\compat\thread
C:\local\tbb\include\tbb\compat\tuple
C:\local\tbb\include\tbb\internal\_aggregator_impl.h
C:\local\tbb\include\tbb\internal\_concurrent_queue_impl.h
C:\local\tbb\include\tbb\internal\_concurrent_unordered_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_async_msg_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_indexer_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_item_buffer_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_join_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_node_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_streaming_node.h
C:\local\tbb\include\tbb\internal\_flow_graph_tagged_buffer_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_trace_impl.h
C:\local\tbb\include\tbb\internal\_flow_graph_types_impl.h
C:\local\tbb\include\tbb\internal\_mutex_padding.h
C:\local\tbb\include\tbb\internal\_range_iterator.h
C:\local\tbb\include\tbb\internal\_tbb_hash_compare_impl.h
C:\local\tbb\include\tbb\internal\_tbb_strings.h
C:\local\tbb\include\tbb\internal\_tbb_windef.h
C:\local\tbb\include\tbb\internal\_template_helpers.h
C:\local\tbb\include\tbb\internal\_x86_eliding_mutex_impl.h
C:\local\tbb\include\tbb\internal\_x86_rtm_rw_mutex_impl.h
C:\local\tbb\include\tbb\machine\gcc_armv7.h
C:\local\tbb\include\tbb\machine\gcc_generic.h
C:\local\tbb\include\tbb\machine\gcc_ia32_common.h
C:\local\tbb\include\tbb\machine\gcc_itsx.h
C:\local\tbb\include\tbb\machine\ibm_aix51.h
C:\local\tbb\include\tbb\machine\icc_generic.h
C:\local\tbb\include\tbb\machine\linux_common.h
C:\local\tbb\include\tbb\machine\linux_ia32.h
C:\local\tbb\include\tbb\machine\linux_ia64.h
C:\local\tbb\include\tbb\machine\linux_intel64.h
C:\local\tbb\include\tbb\machine\macos_common.h
C:\local\tbb\include\tbb\machine\mac_ppc.h
C:\local\tbb\include\tbb\machine\mic_common.h
C:\local\tbb\include\tbb\machine\msvc_armv7.h
C:\local\tbb\include\tbb\machine\msvc_ia32_common.h
C:\local\tbb\include\tbb\machine\sunos_sparc.h
C:\local\tbb\include\tbb\machine\windows_api.h
C:\local\tbb\include\tbb\machine\windows_ia32.h
C:\local\tbb\include\tbb\machine\windows_intel64.h

Does the same apply to your setup, or does your setup diverge from what I
have described? Likely I need to expand the documentation.

dobrzann, 2017-05-11:
Exactly the same configuration and content of the tbb folder.

bubnikv, 2017-05-11:
Do you have another compiler (mingw, gcc, whatever) in your system path? Do
you have another perl in your system path?

For the development build, I have the following environment set up:

set PATH=%PATH%;c:\wperl64d\site\bin;c:\wperl64d\bin
set PERL_CPANM_HOME=c:\wperl64d\cpanm

set BOOST_DIR=C:\local\boost_1_63_0
set BOOST_INCLUDEDIR=%BOOST_DIR%\
set BOOST_LIBRARYDIR=%BOOST_DIR%\lib64-msvc-12.0

set TBB_DIR=C:\local\tbb
set TBB_INCLUDEDIR=%TBB_DIR%\include
set TBB_LIBRARYDIR=%TBB_DIR%\lib64-msvc-12.0\Debug

rem Static link boost libraries
set SLIC3R_STATIC=1
set SLIC3R_HAS_BROKEN_CROAK=1

"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\vcvarsall.bat" amd64

dobrzann, 2017-05-11:
Now I have this:
[CMD.txt](https://github.com/prusa3d/Slic3r/files/992412/CMD.txt)

Can you help me in this case?

dobrzann, 2017-05-21:
Can anyone tell me what it is?
I do everything according to the instructions.

bubnikv, 2017-05-22:
> link @"buildtmp\XS.lds" -out:"blib\arch\auto\Slic3r\XS\XS.dll"
> LINK : fatal error LNK1104: cannot open file 'C:\local\tbb\lib64-msvc-12.0\Debug/tbb_static.lib'

Do you have the following file where the linker looks for it?
'C:\local\tbb\lib64-msvc-12.0\Debug/tbb_static.lib'

Why is there a forward slash before tbb_static.lib?

dobrzann, 2017-05-22:
All files are in place, I don't know why a forward slash is before tbb_static.lib file

dobrzann, 2017-05-23:
problem was solved.

I changed C:\local\tbb\lib64-msvc-12.0\Debug to 'C:\local\tbb\lib64-msvc-12.0\RelWithDebInfo

closed 2017-05-23 by dobrzann