prusa3d/PrusaSlicer #296: Support floating in air over brim

macdarren, 2017-05-12:
### Version
1.34.1 Prusa Edition

### Operating system type + version
OSX 10.12

### Behavior
* _Describe the problem_

support floating in air over brim

* _Steps needed to reproduce the problem_

Brim and any support options (generally just the defaults)
* _Expected Results_

Support extends down to brim

* _Actual Results_

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[image]
[image]
[Floating Support.zip](https://github.com/prusa3d/Slic3r/files/994747/Floating.Support.zip)

bubnikv, 2017-05-20:
I believe this is duplicate of https://github.com/prusa3d/Slic3r/issues/196

bubnikv, 2021-02-22:
Likely fixed with 3985d7bb81ddd542612ec2e8704622be4e90f074
For this model there should be no supports generated below bridges.

closed 2021-02-22 by bubnikv