prusa3d/PrusaSlicer #297: Bad implementation of Rotation/Zoom of Field Of View (FOV) by mouse

Nagymadar, 2017-05-12:
### Version
Slic3r 1.34.1 - Prusa Edition

### Operating system type + version
Win7 64bit

### Behavior
Rotation/Zoom of Field Of View (FOV) by mouse is not user friendly. 
The problem: now the rotation center is in the Z=0 plane, XY= selectable by the projection of the XY coordinates to the Z plane by right mouse button. This is bad. Because if you want to see your object from almost the Z plane (checking first layer in 3D for example) and you shift the FOV by left mouse button just a millimeter... Then the XY rotation center-point will be meters away. After that the smallest rotation by left mouse button moves out the object from the screen and you have to reset your view.

A more user friendly implementation:
Rotation (left mouse button): rotate the FOV around the center of the last selected object. Makes sense, as this is what you are looking at.
Zoom (mouse wheel): quite standard in softwares to zoom using the mouse pointer location as center-point of the zoom.

_Is this a new feature request?_
No.

bubnikv, 2017-06-14:
> The problem: now the rotation center is in the Z=0 plane, XY= selectable by the projection of the XY coordinates to the Z plane by right mouse button.

That is not quite correct. The center of rotation is the center of a bounding box around all objects of the scene.

> This is bad. Because if you want to see your object from almost the Z plane (checking first layer in 3D for example) and you shift the FOV by left mouse button just a millimeter... Then the XY [...]

I am not claiming that it is perfect, but I believe I have fixed some camera issues, some of them introduced by myself and some of them were long standing issues. Would you please retest with the latest release and give me your thoughts?

https://github.com/prusa3d/Slic3r/releases

> A more user friendly implementation: Rotation (left mouse button): rotate the FOV around the center of the last selected object. Makes sense, as this is what you are looking at.

One really does not know whether it was a good idea until one implements it and plays with it.

> Zoom (mouse wheel): quite standard in softwares to zoom using the mouse pointer location as center-point of the zoom.

I have seen it somewhere and I liked it, but I don't think it is a deal breaker.

Do you know some free application I can play with, which implements the scene manipulation the way you described it?

bubnikv, 2017-06-26:
Received no response. The camera handling has been improved significantly in the recent builds. Closing.

closed 2017-06-26 by bubnikv