prusa3d/PrusaSlicer #298: Modifier meshes don't work

Sebastianv650, 2017-05-12:
### Version
1.34.1 Windows 64bit

### Behavior
Modifier meshes seems to be ignored. It seems to be independent from config & modell, therefore I have not uploaded any.
Steps to reproduce:
- Load some modell
- Go into the settings dialog and create a modifier mesh that intersects with the model, for example the complete lower part of it.
- Set the external perimeter speed for the modifier mesh to 10mm/s.
- Save gcode and look for the speeds. The 10mm/s are not used.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
Independent

bubnikv, 2017-05-12:
I believe I remember a similar issue at the upstream slic3r, setting of the velocities is not supported by the modifier meshes.

Sebastianv650, 2017-05-12:
OK, it's also no rear problem. At least, if we get #242 some day - that's the case for my test with modifier mesh :)

But maybe to minimize confused users creating issues, it would be possible to limit the drop down menu from modifier meshed to valid options?

jhitesma, 2017-05-12:
This reminded me of issue #290 which I had been meaning to test.   I just did so and as you can see in my comment on that issue it does appear something has broken with modifier meshes since 1.33.5 as trying the slice the same setup with 1.34.0 I'm getting the modified infill over the entire print not just confined to the modifier mesh like it was with 1.33.5

bubnikv, 2017-07-14:
There was actually a bug in Slic3r specific exactly to the parameter @Sebastianv650 tested: Slic3r did not apply the external perimeter speed, when applied to a modifier mesh.

I will fix that, but even after that change you will not get exactly what you would expect it to do:

This is a screenshot  from Simplify3D, showing the speed by color. You see that there was a modifier mesh applied, where the external perimeters are extruded slowly. But instead of cutting a single perimeter to two pieces, it extrudes two perimeters per layer.

[image]

bubnikv, 2017-07-14:
external_perimeter_speed shall be respected for modifier meshes. Fixed with 
https://github.com/prusa3d/Slic3r/commit/ed73f0b6efdca99a6bbf0d13b0d7b88e4d831d6d

closed 2017-07-14 by bubnikv