prusa3d/PrusaSlicer #299: Strange overextrusion issues with latest release.

MidanRayam, 2017-05-13:
### Version
Version 1.34.1

### Operating system type + version
Windows 10

### Behavior
Look at this images of benchy:
http://imgur.com/a/FVzRL
http://imgur.com/a/KG1VG
It looks like massive overextrusion(it also makes the extruder stepper skip).

I also had some "bulging" issues on the hull of the boat.
http://imgur.com/a/CEAzk

I tried slic3r from http://slic3r.org/ which is working fine with the exact same settings. 

I'm using a wanhao i3 with genuine E3D bowden hotend. 

#### STL/Config (.ZIP) where problem occurs

Here is the config bundle I was using and the actual GCODE:

I'm not sure If this issue is introduced in the latest release, but
[files.zip](https://github.com/prusa3d/Slic3r/files/998768/files.zip)
 the issues started very suddenly for me.

Hugoz12, 2017-05-14:
You use 100mm/s retration speed which may be a bit too high.
Are you sure you use the same setting with the vanilla slic3r? In the config there are two settings maybe you have them switched.

bubnikv, 2017-05-14:
Our retraction settings were tuned to our Prusa i3 mk2 printer and its firmware. I am not sure from the top of my head, but it is possible, that the retraction speed is limited by the firmware.

MidanRayam, 2017-05-14:
I also tried 40 and 50 mm/s retraction speed. But the result was the same. I imported the same config bundle in to vanilla slic3r so the settings should be exactly the same. Do you guys develop the prusa slic3r with other printers in mind or are you guys tuning it for the prusa mk2 without taking other printers into account?

theopensourcerer, 2017-05-22:
I think I am seeing a similar issue. No amount of changing retraction speed or amount seems to make much (any) difference. I will do some further comparisons but wanted to be sure I didn't forget ;-)

My printer is a "clone"; an Anet A8.

bubnikv, 2017-06-05:
> I tried slic3r from http://slic3r.org/ which is working fine with the exact same settings.

Which version? 1.2.9?

> Do you guys develop the prusa slic3r with other printers in mind or are you guys tuning it for the prusa mk2 without taking other printers into account?

We put our customers first, but we are happy that our work is used by the other printer owners as well.

I see you left many parameters to be chosen by Slic3r.
[image]

We do not recommend that. You will be safer, if you start with our settings and modify them as needed.

https://github.com/prusa3d/Slic3r-settings

bubnikv, 2017-06-05:
For some reason Slic3r goes wild with the extrusion speeds with your settings.

[image]

This is how it looks like with our settings:

[image]

It may well be, that the over extrusions and bulging are caused by the excessive nozzle pressure due to an extreme slow down. Simply, the material, which has not been extruded during the quick phase due to a high viscosity of the molten plastic will be extruded during the slow print phase.

MidanRayam, 2017-07-03:
I went from  bowden to a direct drive extruder and the issue is resolved. So @bubnikv might be right with his assumption, since excessive nozzle pressure is more likely with a bowden setup(I think). 

I'm closing this issue.

closed 2017-07-03 by MidanRayam