prusa3d/PrusaSlicer #300: Bridge perimeters printed with perimeter speed

perkmeister, 2017-05-13:
### Version
1.34.1-prus3d-win64

### Operating system type + version
Windows 7 Pro 64 bit

### Behavior
Perimeters are not detected as bridges and are printed at perimeter speeds. This is the same behaviour as reported in #657 in the Slic3r repository which was closed as being fixed, but unfortunately is still alive (possible regression). It is particularly bad for PETG as these need to be printed at low speeds with bridges

If you try to print the attached the perimeters around the bridges in the middle of the object get printed first at high speed, whereas the bridges section will be at the correct bridging speed. This will result in a messy goo on your print with PETG. It appears Slic3r is not recognising these perimeters as needing to be printed at bridge speeds, although it does appear some outer perimeters are detected correctly.

Note: I have also raised a bug report in the Slic3r github, I believe these are separate forks so need one here too (?)

#### STL/Config (.ZIP) where problem occurs
[slc3r_prussa_bridging.zip](https://github.com/prusa3d/Slic3r/files/998896/slc3r_prussa_bridging.zip)

bubnikv, 2017-05-14:
I just recently rewrote the G-code generator from Perl to C++, so I have an
intimate knowledge of the part of the software responsible for this
behavior.

The bridging perimeters are indeed extruded differently than the infill
bridges and currently only the extrusion rate is changed for the bridging
part of the perimeters.

The issue of bridging perimeter speed is a bit more complex. Your request
is basically a limiting case of

https://github.com/prusa3d/Slic3r/issues/284
https://github.com/prusa3d/Slic3r/issues/242
https://github.com/prusa3d/Slic3r/issues/110

I should not just apply the bridging speed for any overhang, but somehow
interpolate these values based on the overhang distance.

perkmeister, 2017-05-14:
OK, thanks. Given you can detect bridging parameters as you're already changing the extrusion width for them, could there be an experimental switch that also changes the speed to those of bridges (and of course turns the fan on)? That sounds like a relatively quick fix, it may at least provide a get out of jail card for some people.

bubnikv, 2017-05-14:
Currently a piece of perimeter is marked as bridging perimeter if it is
supported by less than half the extrusion width. And you probably don't
want all these bridging perimeters to be extruded slowly.

perkmeister, 2017-05-14:
Are you saying that these detected bridging perimeters are extruded though with the bridging extrusion multiplier applied? I'm a bit confused why you would do that and not apply other bridging criteria. Does this mean that perimeter overhangs of say 45 degrees are under-extruded? Clearly this is not as simple as I might think though...

theopensourcerer, 2017-09-27:
I think I am seeing something rather similar, or at least related to this issue... I am running 1.37.1 on Ubuntu 17.04.

See the two attached images showing a couple of layers. I need to bridge over the bottom hole (but retaining a central hole), but the gcode from Slic3r is telling the printer to print the _internal_ perimeter of the centre hole before laying down any infill/bridge material.

layer at Z=2.8mm
[image]

layer at Z=3.2mm
[image]

I then tried setting the "Infill before perimeters" option - which affects the entire print. It did then try to lay down some bridge material but not enough to support the internal perimeters.

Sorry for the rubbish image but obviously hand moved - but I think you can see what I mean.
[image]

espr14, 2019-12-11:
@theopensourcerer, you need to change your model so the few first overhang layers are straight. Look at [https://www.thingiverse.com/thing:3968375](https://www.thingiverse.com/thing:3968375). It will create straight bridges and then it starts to print circle. In most cases it is usable solution.

theopensourcerer, 2019-12-11:
@espr14 That's a really neat idea. Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]