prusa3d/PrusaSlicer #302: Layers omissed during slicing

apphead, 2017-05-15:
### Version
1.34.1

### Operating system type + version
W8.1x64 Pro

### Behavior
Some layers are missing in  2D, Pre-, Layers view, but not in 3D. First and foremost, also the generated gcode shows the same gap.....
[image]

`; Layer 66
G1 Z10.150 F9000.000
; Layer 67
G1 Z10.300 F9000.000
; Layer 68
G1 Z10.450 F9000.000
; Layer 69
G1 Z10.600 F9000.000
; Layer 70
G1 Z10.750 F9000.000
M106 S76.5
; Layer 71
G1 Z10.900 F9000.000`

#### STL/Config (.ZIP) where problem occurs
[spoonrider_buddha.zip](https://github.com/prusa3d/Slic3r/files/1002645/spoonrider_buddha.zip)

bubnikv, 2017-05-16:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/269 and others marked with a "model import" tag. This is an annoying issue, which should be improved soon.

Actually most likely your model is not valid, but other slicers (cura, S3D) often do a better job processing these invalid models.

apphead, 2017-05-16:
That is what I suspected as well. And indeed, the stl I attached has 46 holes according to netfabb. I thought I had tested it with the repaired stl as well (with the same errors), but then I must have made a mistake. 
For reference, I attached the repaired stl which is sliced without the artifacts.
Case closed
[spoonrider_buddha (repaired).zip](https://github.com/prusa3d/Slic3r/files/1003970/spoonrider_buddha.repaired.zip)

closed 2017-05-16 by apphead

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.