prusa3d/PrusaSlicer #304: Make the Build.PL install perl modules in local.lib as in Slic3r original

onekk, 2017-05-17:
Please use the same system used by the original slic3r, in your built the compiled perl modules are written in /home/user/perl5.

In the original Slic3r it build they are written in Slic3r/local-lib/ 

That seems less invading and compatible with multiple version (Think, stable and a devel version installed on same system ).

Best Regards 
Carlo D.

onekk, 2017-05-17:
In the compilation there where some issues, I have modified the Build.pl to reflect those in the original slic3r and added a line in slic3r.pl  to read 

 `BEGIN {
    use FindBin;
    use lib "$FindBin::Bin/lib";
    use local::lib '--no-create', "$FindBin::Bin/local-lib";
}`

The compilation went fine and slic3r-prusa is appearing to work.

More test to come as i'm awaiting for a working printer.

bubnikv, 2017-06-06:
Sorry, but this has a low priority for us. We will rather invest effort into getting rid of Perl instead of trying to patch it. If you want, you may provide a pull request though.

closed 2017-06-06 by bubnikv