prusa3d/PrusaSlicer #305: [Request] Fan Delay

dedalodaelus, 2017-05-19:
### Version
_Version of Slic3r used goes here_
1.3.4.1 stable
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 pro 64bits
### Behavior
* _Describe the problem_
I noticed that the layer fan takes about between 0.5 to 1.5 seconds to achieve full speed. I realized that the overhang or bridging start without the fan activated (this 0.5, to 1.5 seconds), There is a way to take this in count?
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
yes

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-06-08:
This is usually handled by a short full throttle activation of the fan, preferably by the firmware.
This has been discussed recently here: https://github.com/prusa3d/Slic3r/issues/327

Can't you enable the fan blip in your firmware? AFAIK Marlin supports this.

bubnikv, 2017-10-03:
no response. We will have a fan blip enabled in our upcoming firmware.

closed 2017-10-03 by bubnikv

dewi-ny-je, 2020-05-11:
I have fan blip active in my firmware (Klipper) which works well to spin up the fan as fast as possible, but still it takes time and short bridges don't make it: the fan stops before it reaches proper speed.

I propose a setting in the cooling settings which is "spin-up fan time" where the slicer simply shifts up in the final gcode the fan activation by an (estimated) amount of X seconds, based on the travel speed. It won't be perfect but it should be accurate enough.
Quality of bridges would increase.

This is not related to blipping the fan speed, which has other goals: no fan blip can remove the inertia of the fan and of the air mass.

Can we reopen the ticket?