prusa3d/PrusaSlicer #306: internal bridging or overhang something no fan

dedalodaelus, 2017-05-19:
### Version
_Version of Slic3r used goes here_
1.3.4.1
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 pro 64bits
### Behavior
* _Describe the problem_
I noticed that it the internal overhang or bridging sometimes is done without the fan. I saw the plastico drop into the void of the infill.
 See the zip file with STL, GCO, settings and screenshot around more or less when it doing and which zone.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
no
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[issue-overhang-no-fan.zip](https://github.com/prusa3d/Slic3r/files/1012763/issue-overhang-no-fan.zip)

bubnikv, 2017-06-08:
> I noticed that it the internal overhang or bridging sometimes is done without the fan. I saw the plastico drop into the void of the infill.

Would you please be more specific about what you expect Slic3r to do, what is wrong and how it should be improved? Thanks.

bubnikv, 2017-10-03:
no response, closing.

closed 2017-10-03 by bubnikv