prusa3d/PrusaSlicer #309: "Cut" action causes floating prints.

HikariNoKitsune, 2017-05-20:
### Version
1.34.1.24-prusa3d-win64

### Operating system type + version
Windows 10 64bit

### Behavior
I have a model that contains two parts apparently. When I add the file into Silc3r, perform a 'split', delete the unneeded part, then try to cut the remaining part so that it, in theory, lays flat on the build plate without the weird little tab sticking down, it ends up printing the part ~8mm above the build plate! Right where the part itself was 'cut'... I've had this issue before with the normal Slic3r a looooong while ago (nearly 1 1/2-2 years ago when i first started printing). The prusa verions never seemed to do this which is why I've stuck with it.

* _Steps needed to reproduce the problem
[image]
Import multi piece .stl file

[image]
[image]
split into pieces

[image]
[image]
delete pieces that are not needed

[image]
[image]
cut needed piece so it lays flat

[image]
[image]

boom

* _Expected Results_
Part to lay flat on the build plate and actually print at the normal starting height...

* _Actual Results_
Part begins to print ~8mm above the print bed....

#### STL/Config (.ZIP) where problem occurs
[filament_sensorlu_extruder.zip](https://github.com/prusa3d/Slic3r/files/1016543/filament_sensorlu_extruder.zip)

Hugoz12, 2017-05-21:
Can confirm this.
Found a "fix":
- split the parts
- delete the big one
- export the stl
- delete all and add the new stl
- perform cut and slice
I think the issue here is with the removed bigger part giving wrong information when you cut the part. Like it's still using it as a reference for the height as it's not being cut.

HikariNoKitsune, 2017-05-21:
I thought of this too and it only worked maybe 80% of the times I tried it last night. What eventually worked for me every time I tried before just going 'screw it' and heading to bed while printing a different part, was:

- Split the parts
- Delete large one
- Cute the needed piece
- Export cut piece as new stl 
- Delete all pieces
- Import the exported cut piece

That's what worked for me each time I tried it, which was maybe only 5 times cause I was getting to tired.. Hugoz12's fix only seemed to 'usually' work. As in, majority of the time I would do it, it worked, but there was two or three times where it still didn't. Though I think those times were ones I didn't hit arrange to get it into the center of the build plate, but not 100% sure so take it with a grain of salt.

bubnikv, 2017-06-06:
I must be reading the post wrongly, but I cannot reproduce the bug.

1) I load the provided filament_sensorlu_extruder.stl
2) I select the object and press split.
3) I select the small part.
4) I open the cut dialog, enter the 4.5mm cut height, unclick the "keep low part" button, and perform the cut.
5) The smaller component is cut and it is level with the bed.

[image]

Where is the problem?

HikariNoKitsune, 2017-06-06:
It appears to happen when you delete one of the split objects prior to making the cut.

bubnikv, 2017-06-06:
I still cannot reproduce it on Windows 10.

HikariNoKitsune, 2017-06-06:
version 1.34.1.24 correct? I haven't tried it since the update to 1.35.1

bubnikv, 2017-06-06:
I tested with 1.35.1.

HikariNoKitsune, 2017-06-06:
I'll have to redownload the stl and check it out again. Maybe we've gotten lucky and this minor update fixed it! If so, rock on dev team!

bubnikv, 2017-06-06:
Maybe it fixed itself, like my Italian made Fiat.

bubnikv, 2017-06-19:
We at Prusa Research have found one bug, which sound similar:

1) Load a multi-part stl
2) Rotate it by 90 degrees
3) Split it.

Now the split object will jump away from the print bed.

I have fixed this particular issue with https://github.com/prusa3d/Slic3r/commit/7ca02bda0f7ab911a97e435475e05f1b113246d1

bubnikv, 2017-07-07:
No response, similar issue has been fixed with 7ca02bd. Closing.
Feel free to reopen if you reproduce this issue with the upcoming Slic3r PE 1.36.x

closed 2017-07-07 by bubnikv