prusa3d/PrusaSlicer #311: Extra tool moves on each layer since 1.34.1.24

ServAlex, 2017-05-21:
### Version
1.34.1.24-prusa3d-linux64

### Operating system type + version
Antergos (arch linux x64)

### Behavior
On every layer after completing perimeters before starting infill tool moves to X29, Y29 and back to model for no apparent reason.

For example line #168 of attached gcode:
G1 X29.000 Y29.000 F5400.000
This point was far outside of object perimeters.

Removing second extruder from printer settings doesn't help.

Issue didn't occur on 1.34.1 on same config.

Screenshot:
[image]

Config + gcode + stl:
[cubeWithExtraMoves.zip](https://github.com/prusa3d/Slic3r/files/1017072/cubeWithExtraMoves.zip)

bubnikv, 2017-05-22:
It seems to be a bug in the "Avoid crossing perimeters" feature, likely introduced in this build. If I disable the feature, this spurious move goes away.

mrjackson, 2017-05-28:
If you enable Z-Lift it does this move for every lift.

bubnikv, 2017-05-28:
I believe this is a duplicate of 
https://github.com/prusa3d/Slic3r/issues/311
https://github.com/prusa3d/Slic3r/issues/317

bubnikv, 2017-06-02:
Fixed with https://github.com/prusa3d/Slic3r/commit/b5f38dd23f413b06e87ebf005a658bf85bb70af6

closed 2017-06-02 by bubnikv