prusa3d/PrusaSlicer #312: Launch script does not support spaces in directory name

KyleMaas, 2017-05-21:
### Version
1.34.1.24-prusa3d-linux64

### Operating system type + version
Gentoo Linux, although this bug should be the same on any POSIX system.

### Behavior
If you unzip the Slic3r tarball into a directory with a space in it, the launch script does not work properly.  For example, if you have a directory:

/home/foo/3D Printer

And you unzip the Slic3r tarball into it, you get this directory name:

/home/foo/3D Printer/Slic3r-1.34.1.24-prusa3d-linux64-full-201705201439

If you then try to execute (without cd-ing into the directory first - including prompt to clarify exactly what I'm talking about):

foo@localhost:~$ /home/foo/3D\ Printer/Slic3r-1.34.1.24-prusa3d-linux64-full-201705201439/slic3r

It crashes with the following error:

3D Printer/Slic3r/Slic3r-1.34.1.24-prusa3d-linux64-full-201705201439/slic3r: line 6: /home/foo/.
Printer/Slic3r/Slic3r-1.34.1.24-prusa3d-linux64-full-201705201439/bin/perl5.22.0: No such file or directory

#### STL/Config (.ZIP) where problem occurs
None needed.  Script error.

bubnikv, 2017-06-05:
Thanks, the next build will contain an amended start script.
Namely, there was an issue with the 2nd line of the slic3r start script:
it is currently
DIR="$(dirname $0)"
it should be
DIR=$(dirname "$0")

closed 2017-06-05 by bubnikv

bubnikv, 2017-06-05:
Interestingly enough, I had it already correct for OSX.