prusa3d/PrusaSlicer #313: Segmentation fault when opening multiple STL's via CLI

nebbian, 2017-05-22:
### Version
1.34.1.24-prusa3d

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* When loading multiple STL files via the command line interface, the GUI loads, and the first stl is loaded, but then the window closes and the CLI reports: Segmentation fault: 11

* _Steps needed to reproduce the problem_
Enter the following on the command line:
`/Applications/Slic3r.app/Contents/MacOS/Slic3r --gui -o test.gcode 15x15x15.stl 15x15x0.5.stl`

* _Expected Results_
Both models show up on the plater.
* _Actual Results_
Error message is displayed in the CLI, and the window disappears.

You can stop the error by using the --no-plater option in the CLI, but this defeats the purpose of loading multiple files into the plater.

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
[slic3r problem.zip](https://github.com/prusa3d/Slic3r/files/1018036/slic3r.problem.zip)

bubnikv, 2017-07-11:
Following commit https://github.com/prusa3d/Slic3r/commit/32213ce6798606df405cc6106b9c95106d253090 seems to patch this crash by using just a single notification dialog for loading all the STLs.

If you encounter the issue with the upcoming 1.36.x, please feel free to reopen.

closed 2017-07-11 by bubnikv

nebbian, 2017-07-11:
Thanks @bubnikv, I'm looking forward to testing this out.  This fix will really help [this project](https://github.com/nebbian/RoboxSlicerExtension).

nebbian, 2017-07-18:
Hey @bubnikv, just letting you know that I was eagerly awaiting the next release of slic3r due to handling multiple files properly, and it didn't disappoint!  It works perfectly.  I can put multiple files into the vendor's software, and they transfer to Slic3r properly.  They even maintain their relative positions, which I wasn't expecting!

Just letting you know that your efforts are very much appreciated.  Keep up the great work.

bubnikv, 2017-07-18:
Thanks, Ben.

> I can put multiple files into the vendor's software, and they transfer to
Slic3r properly.

I don't understand how that is possible, but the main thing is, you are
happy :-)

Vojtech