prusa3d/PrusaSlicer #314: Raft and support dont work together

jindrichbenes, 2017-05-22:
Raft layers generate first, then does the support.

[image]

[image]

[image]

jindrichbenes, 2017-05-22:
[wrong raft and support.zip](https://github.com/prusa3d/Slic3r/files/1018397/wrong.raft.and.support.zip)

bubnikv, 2017-06-08:
Would you please attach the STL?
Is this a general issue, reprodducible with any object?

jindrichbenes, 2017-06-09:
general issue, maybe MM general related. Try these .AMFs

[wrong rar amfs.zip](https://github.com/prusa3d/Slic3r/files/1064531/wrong.rar.amfs.zip)

bubnikv, 2017-06-13:
There was an issue with raft & soluble support.
Also there was a bug, where the support was not generated correctly
after a change of the support Z gap, one had to delete the object and reload it to get the supports generated correctly after the change of the support Z gap.

Fixed with https://github.com/prusa3d/Slic3r/commit/8a2a9abbd48dcc6c3f11dc9445df32045d71f47a

closed 2017-06-13 by bubnikv