prusa3d/PrusaSlicer #315: Clicking on "Layer Editing" causes crash

FHeilmann, 2017-05-22:
### Version
_Version of Slic3r used goes here_
1.34.1.24-prusa3d-win64
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Pro x64
### Behavior
* _Describe the problem_
Clicking on "Layer Editing" causes a crash
* _Steps needed to reproduce the problem_
Load g-code file
Press "Layer Editing"
* _Expected Results_
"Layer Editing" starts
* _Actual Results_
Slic3r crashes
* _Console Output_
```
Slic3r installed in a loclized path. Using an 8.3 path: "C:\PROGRA~1\SLIC3R~1\SLIC3R~1.1-P\"
Info about pegboard_updated_20170522-9233-qnfax7-0.stl:
  size:              x=33.950 y=19.500 z=22.773
  number of facets:  2380
  number of shells:  3
  volume:            5777.922
  needed repair:     no
Illegal division by zero at C:/PROGRA~1/SLIC3R~1/SLIC3R~1.1-P/lib/Slic3r/GUI/3DScene.pm line 1478.
```

_Is this a new feature request?_
No
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[slic3r.zip](https://github.com/prusa3d/Slic3r/files/1019029/slic3r.zip)

nebbian, 2017-05-23:
I noticed this today on Mac OS X 10.10.5 as well.

beast-CAT, 2017-05-31:
I'm having the same problem on Windows 7 Pro 64bit.

BirkBinnard, 2017-06-03:
Yup - happens on Win10-64 also

bubnikv, 2017-06-05:
This has been fixed with https://github.com/prusa3d/Slic3r/commit/88c9ae6ca690e8661b08bf31b14dfec98af7e442
The problem was in the default value "0" for the max_layer_height, which did not translate to 3/4 of the nozzle diameter correctly at the user interface level, causing a division by zero.

closed 2017-06-05 by bubnikv