prusa3d/PrusaSlicer #316: Supports go through object

jindrichbenes, 2017-05-25:
ver. 1.35.0.7

two separate objects

[image]

bubnikv, 2017-07-07:
Supports are generated per object, and collisions of objects are not tested by Slic3r. This is a known issue.

ciekawy, 2022-05-10:
why not combine merge objects into one first and only then slice, generate support, raft etc

redfast00, 2023-02-10:
@lukasmatena I just came across this issue: as an easier alternative to automatically fixing this, would it be possible to detect overlapping gcode and show a warning 'The gcode is overlapping. Merge the models to fix this'? I just had to scrap a large part because of this issue.

morphias2004, 2024-05-10:
+1

7 years on and still an issue. :(

el-bart, 2024-10-05:
there are 2 more "interesting" side effects / fallouts of this bug, namely:
* when using `--duplicate` cmd line option objects are auto-positioned incorrectly, causing collision with supports
* when running from cmd line, slicing ends up with SEGV

see #13464 (closed as duplicate) for more details.

Tupson444, 2025-05-28:
The workaround to merge objects into one is no longer a good option because cancel object wouldn't work with it - you won't be able to cancel just the failed object because they are all treated as one.

morphias2004, 2026-09-18:
> +1
> 
> 7 years on and still an issue. :(

Nearly 9.5 years on and Prusa just close this case because PS3 is coming. Is it fixed in there?