prusa3d/PrusaSlicer #317: Avoid crossing perimeters is broken in 1.34.1.24

maso27, 2017-05-25:
### Version
1.34.1.24

### Operating system type + version
Windows 7 Professional x64

### Behavior
When building with "avoid crossing perimeters" enabled, the head travels to multiple points outside of the envelope of the print.  In some cases it departs from the print significantly.
The problem is not present under version 1.33.8.

* _Steps needed to reproduce the problem_
Slice a part with all boxes checked under "Quality (slower slicing)".  This might work with different settings, but that is what I've used.

* _Expected Results_
(I used Repetier-Host for the display because I couldn't figure out how to display travel with slic3r preview):
**Results with Slic3r v1.33.8**
[image]

* _Actual Results_
**Results with Slic3r v1.34.1.24, with the same settings**
[image]

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[STL_and_config.zip](https://github.com/prusa3d/Slic3r/files/1029478/STL_and_config.zip)

Hugoz12, 2017-05-25:
Dupicate of #311

bubnikv, 2017-06-02:
Fixed with https://github.com/prusa3d/Slic3r/commit/b5f38dd23f413b06e87ebf005a658bf85bb70af6

closed 2017-06-02 by bubnikv