prusa3d/PrusaSlicer #319: [Request]G-code on per filament setting

sarf2k4, 2017-05-26:
Would like to ask some validation whether or not this is feasible because on the recent slic3r release, the linear advanced has been removed in the printer section. 

I read several topics regarding the linear advanced where you would adjust the K values to maintain constant pressure inside the hot end. 

To my knowledge from reading those topics, it is different from material type and even if both abs material also might have different K values.

What's more is that printing with different material also gives a possibilities that the user has to adjust the z-offset while printing. With this, hopefully enables the user to add a z-offset change g-code without to fiddle from the lcd panel anymore, provided that the user knows the correct settings for a good layer height.

Sebastianv650, 2017-05-26:
I don't believe two ABS materials will have different K values. If both are realy ABS and not some alloy, the spring constant (and K is nothing else in the end) will be the same.
Nevertheless, you are right when you say seperate start gcode sections linked to the selected filament would be helpful to set LIN_ADVANCE K and simmilar. I also made the experience that PLA needs another z offset than PETG, which I guess comes frome the different printing temperatures and therefore different expansion of the hot end and bed due to heat.

sarf2k4, 2017-05-26:
What i mean is same abs material but different manufacturer, i dont know if
i stated that already, replying in email though.

As of now, i dont think the z-offset value can be set via offset instead of
absolute value

sarf2k4, 2017-05-26:
I also opened another threads in prusa firmwaee and marlin firmware
regarding this

tcm0116, 2017-05-26:
I submitted a similar request in #265, but there's been no discussion on it.

sarf2k4, 2017-05-26:
@tcm0116 , maybe your description not catchy enough? ;P

I guess another workaround is to use the post processor but not all of us know how to use it including me nor the right software to use it either

fiveangle, 2017-05-27:
I agree with this - if only so we can set a different Z offset: PETG indeed requires very different offset that PLA, and it is tedious to try and guestimate the amount required in the livestepping.

bubnikv, 2017-06-05:
Implemented with https://github.com/prusa3d/Slic3r/commit/ad3be1a69e6cf8ec8d3169d0b346b664040b2c3b

Based on the implementation by @lordofhyphens
https://github.com/alexrj/Slic3r/commit/19eea19d916d2b45ca45e22cf34e4e148ac477a1

closed 2017-06-05 by bubnikv