prusa3d/PrusaSlicer #320: [Request] Auto arrange object around the build plate to minimize wear and tear of the PEI in the middle spot

oleost, 2017-05-26:
### Behavior
* _The PEI sheet gets weardown in the middle due to its the default placement of prints, also looking at pictures the place where the screws are seems like a weak point of the PEI._
* _Expected Results:_ Place objects random on built plate to minimize wear in the middle.
* _Actual Results:_ Unnecessary wear of the built plate since the object are always in middle

_Is this a new feature request?_ Yes

bubnikv, 2017-06-02:
I understand your reasons, but I am a bit opposed of implementing such a feature as it adds complexity to the software and its configuration.

Do you have a proposal of the amount of jitter the software should introduce to the object placement? Just a couple of millimeters, or should it place the object at a completely different spot on the bed?

I am actually more inclined to another proposal: Allow mapping of a custom bitmap over the bed preview. This would allow you to visualize scratches, holes and other blemish of the PEI sheet.

oleost, 2017-06-02:
I would say stay within 85% of the build platform, and then random placement. ( As people are using different printers etc it would be to much and little gain to track overall usage of build platform )

I believe example the Formlabs Form 2 does this actually, but its also gets wear down quicker. Unsure if its in printer firmware or slicer. But for our needs it would be easiest and probably best to just have the slicer place it random, and of course user configurable to turn this feature on.

yoyo42, 2018-12-28:
I was just about to post the same request for the same reason. My PEI plate wears in the center and I always have to move things around myself. I would be happy for the objects to be anywhere on the build plate as long as any brim/skirt/support still fits in.

jonnieZG, 2019-04-14:
I would also love to have that option. Here is my proposal:

* There should be an option, that when enabled, would randomly place objects within the print bed in order to level bed-wearing.
* Auto-Arranging of multiple objects would be performed the same way as it is now, but around a randomly selected point on the printbed, rather than its center (as currently is).
* Instead of "Auto-center Parts" checkbox there should be a select-one-menu called "Object positioning" with the following options:

1. Centered on Load (manual repositioning allowed)
2. Randomly positioned (manual repositioning allowed)

haarp, 2019-07-07:
This is a good idea. Additionally, I'd like to propose the ability to define zones that the positioning avoids, in case an user has some damage on their PEI sheet in that location.

JohnOCFII, 2019-07-07:
> Additionally, I'd like to propose the ability to define zones that the positioning avoids, in case an user has some damage on their PEI sheet in that location.

I would like this option as well -- and it seems like it would be easier to implement the random placement with exclusion areas.

rongith, 2019-09-20:
Arranging parts with a bigger distance would be nice when printing few small parts that has some difficulties to dissipate the heat of 260-280°C filaments. Then the hotend wouldn't radiate over them all the time.

bubnikv, 2019-09-23:
I would rather see the jitter to be implemented inside the firmware.
Imagine printing one G-code over and over. To minimize wear and tear of the
PEI sheet, the firmware would need to randomize the offset.

haarp, 2019-09-23:
For that the firmware would have to determine the total size und location of the print area used for each print, i.e. look ahead through the entire gcode. With that puny 8-bit CPU, I don't think that's reasonable. Also won't work when printing over USB or Octoprint.

bubnikv, 2019-09-23:
My proposal is to export the bounding box into the G-code by slicer.

yoyo42, 2020-02-02:
I can see that as a separate feature request for the firmware but this was more about the behaviour of PrusaSlicer when importing one or more objects. Having the firmware able to move things around but relying on the slicer exporting a bounding box means that _every_ slicer needs to cooperate _and_ the firmware needs to be changed. That's a lot to change across multiple projects, when it could be solved purely in PrusaSlicer. If someone wants to print the same model over and over there's nothing to stop them reslicing it in a different position . I suspect that's a less widespread use case for most hobby users at least than doing different models each time.

3Delights, 2021-04-01:
It would actually need to be pseudo random as in a truly random system the same place could (and probably would) get chosen several times in a row, thus defeating the whole point of the feature!

Why don't people just move the object in PrusaSlicer to a different part of the bed before slicing?

ctag, 2021-04-01:
I don't think there's a need for a distinction. Even if a truly random generator picks the same spot 10 or 20 or 30 times, that won't cause the kind of wear that hundreds of prints can and do cause.

Moving by hand is fine, but it is a clunky thing to do in the UI. And Auto-arrange is a valuable feature that happens to always pick the same spot.

claytono, 2021-04-05:
@3Delights There are plenty of conveniences that aren't necessary but the slicer has.  For example, why have an auto arrange at all?  People can obviously arrange items themselves, so the slicer doesn't *need* to have that feature.  However, I think most of us appreciate being able to hit the arrange button and getting something relatively close to optimal.  I think this feature falls into the same sort of category.

bubnikv, 2021-04-07:
You can likely already script that using the custom G-code in PrusaSlicer 2.3.

In PrusaSlicer 2.3.0 there is a random(low, high) function.
Also there are print_bed_min, print_bed_max, print_bed_size, first_layer_print_min, first_layer_print_max, first_layer_print_size placeholders, all vectors of two. Together with the min, max functions and additions you should be able to script the random shift of your print.

Caveat: The random function in PrusaSlicer 2.3 is not randomly initialized. We will fix that in PrusaSlicer 2.4.

jonnieZG, 2021-04-07:
> You can likely already script that using the custom G-code in PrusaSlicer 2.3.
>
> In PrusaSlicer 2.3.0 there is a random(low, high) function.
> Also there are print_bed_min, print_bed_max, [...]

Could you please post an example?

ctag, 2021-04-08:
Here's what I was able to come up with just poking around at it. I'm not very familiar with the macros so this might be all wrong.

```
G92 X{100.0 + random( -(print_bed_max[0] - first_layer_print_max[0]), first_layer_print_min[0] - print_bed_min[0] )} ; x-offset
G92 Y{-3.0 + random( -(print_bed_max[1] - first_layer_print_max[1]), first_layer_print_min[1] - print_bed_min[1] )} ; y-offset
```

And in the context of my custom start G-code:

```
M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.9.3 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S{first_layer_temperature[0] - 10}; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}

G92 X{100.0 + random( -(print_bed_max[0] - first_layer_print_max[0]), first_layer_print_min[0] - print_bed_min[0] )} ; x-offset
G92 Y{-3.0 + random( -(print_bed_max[1] - first_layer_print_max[1]), first_layer_print_min[1] - print_bed_min[1] )} ; y-offset

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}
```

Either as a consequence of me doing it wrong, or of the static seed for random(), the parts always get "randomly" placed to the same place, no matter which part. 
[image]
[image]

bubnikv, 2021-04-09:
@ctag

That's exactly what I meant. You are doing it correctly.

The initial G-code preview is not shifted as the G92 is not there yet, but the final G-code preview shows the random offset.

> Either as a consequence of me doing it wrong, or of the static seed for random(), the parts always get "randomly" placed to the same place, no matter which part.

That is a bug in PrusaSlicer 2.3.0, the random is not seeded. This bug will be fixed in PrusaSlicer 2.4.0 and it is actually already fixed on the master.

bubnikv, 2021-04-09:
You also may want to use the max function to limit the randomization.

haarp, 2021-04-10:
Nice approach, but it can fail if the part is larger than just its first layer. Are there any variable that define a bounding box for the entire part instead of just first_layer_print_* ?

haarp, 2021-07-12:
@ctag 

There's one issue with this gcode. When the printer homes after finishing its print, the print head will often smash into the X-end. It seems some coordinates are not being set properly. Not sure what exactly is causing it.

ctag, 2021-07-17:
> @ctag
> 
> There's one issue with this gcode. When the printer homes after finishing its print, the print head will often smash into the X-end. It seems some coordinates are not being set properly. Not sure what exactly is causing it.

Thanks for sharing your experience with it. I'm not sure what's going on there, can you share a .3mf file it happens with?

haarp, 2021-08-14:
> Thanks for sharing your experience with it. I'm not sure what's going on there, can you share a .3mf file it happens with?

Here you go.

[av.zip](https://github.com/prusa3d/PrusaSlicer/files/6986611/av.zip) (rename from .zip to .3mf)

[av_0.4n_0.28mm_PETG_MK3.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/6986612/av_0.4n_0.28mm_PETG_MK3.gcode.zip) and the resulting g-code (extract from .zip)

In this example, the Y axis crashed into the front during the final "show" move.

ctag, 2021-11-28:
Thanks haarp. It looks like since the G92 changes absolute position that the final "G1 X0 Y200 F6000" on line 13719 causes the extruder to crash. So even if the random() worked, this approach may not. 

Speaking of random, I have the beta 2.4.0 compiled, and the random function works now!
[image]

[image]

[image]

haarp, 2021-11-29:
@ctag 

My Prusaslicer is patched, so random() worked to begin with :)

Come to think of it, the problem is tha we set a different home with G92, which then causes the printer to ram into things. I wonder if [G53](https://marlinfw.org/docs/gcode/G053.html) before hte final "present" move would fix that. It resets the printer to the native workspace.

ctag, 2021-11-29:
It looks like even if G53 was supported this solution couldn't be entirely in g-code since only the first layer sizes are exposed to the Custom Start G-code.

New placeholders could be added, such as "print_min" and "print_max" or the feature could be directly added:

[image]

Either way it looks like the slicer C++ would need to be mucked with.

haarp, 2021-11-29:
> t looks like even if G53 was supported this solution couldn't be entirely in g-code since only the first layer sizes are exposed to the Custom Start G-code.

The way I understand G53, it can be used to return to the 0,0,0 coordinate system, undoing any G92 commands prior to it. So it should work fine, if it was supported. G92.1 would be another contender, but the Prusa firmware doesn't recognize it either.

ctag, 2021-11-29:
Right, thank you for the clarification.

What I meant to say is that even if G53 support was added, the G92 commands would also need new parameters (instead of first_layer_print_max) in order to prevent overhangs from being placed outside the build area.

At that point the slicer core has to be edited and accepted. Why not just make wear leveling a native feature instead of making those who want it hunt down custom g-code?

ctag, 2021-12-01:
OK, I cobbled together a minimum-viable demo of that feature. 
https://github.com/ctag/PrusaSlicer

https://www.youtube.com/watch?v=W_EV1MWz0wo

I think the eventual solution could be something as cool as tracking heatmaps of per-plate placement usage, and doing active wear-leveling instead of randomized placement. Of course even that solution wouldn't work for the serialized printing from the same gcode over and over...

haarp, 2021-12-01:
This looks nice! You should make a pull request.

claytono, 2021-12-01:
@ctag that's awesome, thanks for working on it!

ctag, 2021-12-02:
I would love to make this a feature and get it accepted, but it'll take some patience since I'm learning as I go.

Looking back through this thread to refresh my memory. It looks like there are two use cases represented:

1. Cottage industry. Serialized printing of the same gcode file over and over. I expect that the effects of print bed wear are most pronounced in this situation.
2. Varied, one-off use. The printer rarely prints the same gcode file more than a dozen times. This is me, and I assume (with no research) that most MK3 users fall into this camp. I also believe that an easy-to-use PrusaSlicer feature is best served on this group, since cottage industry printers likely already pay more attention to configuring.

The solution I'm thinking of only really helps type 2 users. It looks like user type 1 would be best served with a firmware implementation. Modifying the MK3 firmware to include a "randomize by this margin" comment at the bottom of the gcode (where all the other slicing data is kept) and have it interpreted on each run is feasible. In fact, since the steel sheet profiles are saved on the printer, bundling the randomization there would allow for the "real" wear leveling I mentioned above. But such a feature would in some way go against the standardization of gcode, and I wouldn't expect the Prusa team to accept it as a feature.

And just thinking about this (so I may be wrong), it seems that serialized printing may not benefit much from randomization:

- Printing a large item (only one fits on the print bed). In this situation the center of the print bed is printed on every single print, by definition (excluding complicated first layer shapes). so randomizing the location by a few mm isn't going to save the middle of the plate from accumulating wear. 
- Printing small items (multiple may fit on print bed). Here I would expect many serialized printers to just print the multiple items together, but even if they don't then this could become a handful of gcode files with the part arranged in different places, then the printer is cycled through those files to get some amount of wear leveling.

So we're left with a feature that's going to be implemented in PrusaSlicer and only randomize at the slicing stage, not printing. That could be done in two ways that we've discussed:

1. New gcode parameters that encompass the full part's footprint on the build plate, i.e. 'print_min' and 'print_max'
2. A feature of the Arrange button. There's already precedent for controls and toggles that adjust how Arrange works, this would just be another.

One issue I see that may determine which one of those is feasible: right now it looks like the arrange button and the first_layer_print_max/min do not take into consideration skirts, which can wind up outside the build plate. By auto-placing everything at the center, the parts either fit with their brim or they do not, and so looking at the code it appears to be ignored.

---

> Arranging parts with a bigger distance would be nice when printing few small parts that has some difficulties to dissipate the heat of 260-280°C filaments. Then the hotend wouldn't radiate over them all the time. I would like changing the parameter deciding of the distance between parts.

@rongith It's been a while since this comment was made, but I just want to note: this is available in the current PrusaSlicer!
Right click on the Arrange button, and select the new distance:
[image]
[image]

cnlson, 2022-02-03:
@ctag so i suggested dividing the plate into 4 section 120x105 and filling them sequentially. So the first design is centered in the front left quadrant, then the next sliced design in the next quad. if you need more room spillover to the next quadrant. and so on.

nomike, 2024-03-16:
I wanted to add a few cents as to why such a randomization feature would be good:

Humans are inherently bad at randomizing things. If you ask a person to come up with a random password, they will hit random keys on their keyboard and the probability is quite high that the password includes "asd" in it for example.

Same with randomly placing objects on the build plate: The corner and extreme sides of the build plate will almost never see anything printed on them.

There are some good ideas floating around like proper wear leveling, defining "bad sectors" which are avoided during slicing, etc..

But these seem to be quite complex to implement. The wear leveling requires keeping a record of areas printed upon, which is something which can't even be done properly as the slicer has no idea which things have actually been printed and not just sliced, and also how often this was the case.

Defining bad sectors requires a complicated user interface, the storage of them and it also makes the algorithm quite complex.

So I would vote for implementing the simplest solution first:

An binary option for the auto-arrange feature. When selected this happens:

1. All objects are distributed with the already existing code. No change necessary there.
2. The bounding box of all objects is calculated. The code for that already exists as the slicer calculates this white corner-indicators when all items are selected.
3. The starting y-coordinate is set to rand(0, build_plate[max_y] - bounding_box[y[
4. Same thing for the x-coordinate

Of course this will never be perfect, large objects will always cover the center for example, and so on.

But it doesn't have to be. Instead of never getting a feature which isn't perfect.
Let's start with an easy solution which is feasible to implement and which is already a massive improvement over what we have right now.

One thing to consider though: As mentioned before, the arrange feature doesn't know about tool paths, so maybe this could be an issue.

A possible easy workaround for this would be to define a margin which the algorithm is trying to avoid, e.g. 5 millimeters.
This means that the edges of the print bed will not be used, but it still is massively better than putting all objects in the center all the time.

Tupson444, 2024-03-16:
I see no practical reasons for always arranging in the center (there is no benefit in centering, unless it is really heavy print in which case it would also be big enough so it would have to be centered anyways). So random arranging should even be the default behavior, but adding it just as an option would be good too.

I always manually move objects around the bed to avoid placing them in the same spot. Since large objects (or many small objects) have to cover the center naturally, when I print small things I always place them AWAY from center. Therefore the random placement should always try to avoid the center _(because even that way the center will still suffer the most wear, unless someone never prints anything larger than half the size of the bed)_.

ctag, 2024-03-16:
For those that could use this workaround, I'd like to mention that the GCode in this comment works now. The random function has been updated since I made the comment, and part locations will now be randomized. Be aware that overhangs can still be placed off the print bed due to the variables depending only on first-layer dimensions.
https://github.com/prusa3d/PrusaSlicer/issues/320#issuecomment-815912021

[image]

cnlson, 2024-03-16:
> For those that could use this workaround, I'd like to mention that the GCode in the comment link below works now. The random function has been updated since I made the comment, and part locations will now be randomized. Be aware that overhangs can still be placed off the print bed due to the variables depending only on first-layer dimensions. [#320 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/320#issuecomment-815912021)
> 
well, in this case, the part is larger on the bottom and still hangs off the side
[image]

ctag, 2024-03-16:
Well shoot. @cnlson Can you provide the .3MF file for that project?

cnlson, 2024-03-17:
here you go
zipped so it will attach
[geocache_V2.1.zip](https://github.com/prusa3d/PrusaSlicer/files/14628690/geocache_V2.1.zip)

ctag, 2024-03-17:
> here you go zipped so it will attach [geocache_V2.1.zip](https://github.com/prusa3d/PrusaSlicer/files/14628690/geocache_V2.1.zip)

Thanks! To the best that I can tell, it looks fine. I added some debugging statements that you can see below. What version of PrusaSlicer are you using?

```
M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.11.0 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S140
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Z0.2 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
G92 E0
M221 S{if layer_height<0.075}100{else}95{endif}

M117 First layer X max: {first_layer_print_max[0]}
M117 First layer X min: {first_layer_print_min[0]}
M117 Bed X max: {print_bed_max[0]}
M117 Bed X min: {print_bed_min[0]}

M117 First layer Y max: {first_layer_print_max[1]}
M117 First layer Y min: {first_layer_print_min[1]}
M117 Bed Y max: {print_bed_max[1]}
M117 Bed Y min: {print_bed_min[1]}

M117 X Rand: {-(print_bed_max[0] - first_layer_print_max[0])}, {first_layer_print_min[0] - print_bed_min[0]}
M117 Y Rand: {-(print_bed_max[1] - first_layer_print_max[1])}, {first_layer_print_min[1] - print_bed_min[1]}

G92 X{100.0 + random( -(print_bed_max[0] - first_layer_print_max[0]), first_layer_print_min[0] - print_bed_min[0] )} ; x-offset
G92 Y{-3.0 + random( -(print_bed_max[1] - first_layer_print_max[1]), first_layer_print_min[1] - print_bed_min[1] )} ; y-offset
```

> G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
M73 Q0 S94
G92 E0
M221 S95
> 
> M117 First layer X max: 178.623
M117 First layer X min: 71.3136
M117 Bed X max: 250
M117 Bed X min: 0
> 
> M117 First layer Y max: 144.214
M117 First layer Y min: 65.7222
M117 Bed Y max: 210
M117 Bed Y min: 0
> 
> M117 X Rand: -71.3772, 71.3136
M117 Y Rand: -65.7863, 65.7222
> 
> G92 X127.563 ; x-offset
G92 Y-25.0241 ; y-offset
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K0.04 ; Filament gcode LA 1.5
M900 K24 ; Filament gcode LA 1.0

cnlson, 2024-03-17:
> Thanks! To the best that I can tell, it looks fine. I added some debugging statements that you can see below. What version of PrusaSlicer are you using?
> 
2.7.1 

i only added 

>G92 X{100.0 + random( -(print_bed_max[0] - first_layer_print_max[0]), first_layer_print_min[0] - print_bed_min[0] )} ; x-offset
G92 Y{-3.0 + random( -(print_bed_max[1] - first_layer_print_max[1]), first_layer_print_min[1] - print_bed_min[1] )} ; y-offset

i didn't add 

>M117 First layer X max: {first_layer_print_max[0]}
M117 First layer X min: {first_layer_print_min[0]}
M117 Bed X max: {print_bed_max[0]}
M117 Bed X min: {print_bed_min[0]}

>M117 First layer Y max: {first_layer_print_max[1]}
M117 First layer Y min: {first_layer_print_min[1]}
M117 Bed Y max: {print_bed_max[1]}
M117 Bed Y min: {print_bed_min[1]}

>M117 X Rand: {-(print_bed_max[0] - first_layer_print_max[0])}, {first_layer_print_min[0] - print_bed_min[0]}
M117 Y Rand: {-(print_bed_max[1] - first_layer_print_max[1])}, {first_layer_print_min[1] - print_bed_min[1]}

ctag, 2024-03-17:
Got it, thanks. 

I only added those M117 lines to see what values the templates were generating. If you can get your slicer to make a "bad" slice again, having those might help explain what's going wrong. 

On my prusaslicer 2.7.2 you can follow the numbers from those M117 lines to see it shouldn't be able to relocate the print off of the print bed.

cnlson, 2024-03-18:
> Got it, thanks.
>
> I only added those M117 lines to see what values the templates were generating. If you can get your slicer to make a "bad" slice again, having those might help explain what's going [...]

i updated to 2.7.2 and I created a separate profile for it. so far it seems like it is putting things on the bed. I'll use it on some my my frequent hitters to spread the wear around.

cnlson, 2024-03-27:
> Got it, thanks.
>
> I only added those M117 lines to see what values the templates were generating. If you can get your slicer to make a "bad" slice again, having those might help explain what's going [...]

OK I've been using the random since last week and I had an interesting issue. I had a crash so it went to 0,0 and when it resumed it was off by either a random amount or back to the center.

Is that because it either looses the offset or re-randoms?

nomike, 2024-07-04:
I gave this some more thought. I usually don't print dozens of copies of the same thing. So what I end up doing a lot is to place objects on my build plate in the slicer, use the arrange function to neatly organize them in the center of the plate and then select all of them and move the whole bunch to some random position.

That humans are terrible at randomizing things is something we already established earlier.

So my suggestion is to add a flag to the arrange settings for enabling random placement.
It happens sometimes that you move an object to the edge of the plate, the object still being marked as green, but once you slice you get an error about a toolpath bing outside the build area. For this an additional parameter should be added to allow for a margin around the edges of the build plate.

And then whenever you hit arrange, it will do this:

1. Run the existing arrange algorithm which places everything in the center of the build plate.
2. Calculate the x- and y- size of all objects combined (the slicer does that already when you select everything as it needs to display the white corners of the selection).
3. Calculate X using `x = random(0, plate.max_x - (selection.max_x - selection.min_x) - arrange.margin)` and y using `y = random(0, plate.max_x - (selection.max_x - selection.min_x) - arrange.margin)`.
4. Move the obects to the calculated position.

So each time I put somthing on the plate, I hit the arrange button, the objects get randomly placed somewhere and the plate is worn  out more uniformly.

This is by no means perfect of course and doesn´t  solve the issue for people who print one slice dozens of times over and over, but I assume this will already help >80% of the users.

Tupson444, 2026-09-13:
Still relevant

KlausKraemer, 2026-09-14:
Still relevant for Slicer 3.x.

The print sheets get worn down earlier, if you let Prusa Slicer arrange the objects without intervening manually.