prusa3d/PrusaSlicer #321: [Request]Change per object layer height to something else

sarf2k4, 2017-05-31:
This option in my on opinion, has become useless since the arrival of variable layer height feature and would like to request that this option to be changed to temperature and/or fan instead.

On the current setups, it has z min, z max and layer height, I do remember fiddling with this setting before but kind of tedious that we got to input for both z min and z max to change and restore the layers.

My proposal is that to just use 2 columns; z-start, temperature and fan speed in %. if the temperature were to be left blank but the fan setting are present, it will ignore temperature and not adjusted, and vice versa. except that this setting, will override the default fan settings under "filament settings"

bubnikv, 2017-06-02:
I wonder why do you need to set the fan speed and nozzle (or bed?) temperature per layer range. Would you please expand on that?

sarf2k4, 2017-06-02:
Just the temperature for the hotend will do. In cases of ABS material and big prints, it requires higher temperature. If for example I were to print a big 3dbenchy that were upscaled to max that MK2 able to print, the hull that's about 40' degree angle with this setting 240'c on first, the rest is 260'c might have an impact on this angle at higher temperature. While using fan might solve the curling problem on these angle but might also introduce another issue, delaminations.

Other way to achieve this is of course, to edit the gcode file manually by searching desired "z-start/stop" for temperature change and/or fan speed. But these can be tedious and might disrupt the whole gcode itself.

Or, I just come up with another, "adaptive temperature" on per filament setting, they would have these "min temp" and "max temp". The "adaptive temperature" would be adjusted same way the "adaptive layer height" behaves. This way don't have to change the current existing layer range, just add another like "layer editing" section.

In conclusion, printing an object with a higher temp setting would produce extra structural rigidity but might produce warping. So it is a double edged sword =D

Hope this is helpful =D

m-Th, 2017-06-28:
More complete specs here: https://github.com/prusa3d/Slic3r/issues/378

NTrack1, 2019-01-13:
I suppose an easy way to have per layer temperature/cooling control is to add Before/After Layer Change custom Gcode under Filament settings profiles . Using placeholders is easy enough to have different settings for per layer cooling/ temperatures for a material without having to have duplicates of printer profiles for each material and nozzle.

WoosterInitiative, 2023-06-28:
This is pretty old, and even my addition probably doesn't add much, but I would find this handy *specifically* for things like temp towers. I found someone that did it with "before layer change" gcode changes and I'm considering writing a gcode post-processor that does it a bit more elegantly, but, at least for single-extruder/single-material prints, this seems pretty easy to implement.

Aldekein, 2025-08-01:
After installing 2 upgraded fans on new mounts, I tried to set fan speed per layer range, to experiment with crazy overhangs, and... well, setting is not there :(

closed 2026-09-29 by prusaslicer-bot[bot]