prusa3d/PrusaSlicer #323: Gcode includes moves that travel away from the object being printed, to outside the print area

eanow, 2017-06-01:
### Version
Slic3r 1.34.1.24

### Operating system type + version
MacOS, and Linux

### Behavior
When slicing many models, the resultant gcode includes excursions outside of the area where the objects lie. Specifically, I am slicing for my delta printer, and it will create a travel move which moves to a point several centimeters away from the object.

Attached I've included the stl, config file, and one of the gcode files I generated. On several layers (including 38) there is a move to a location far outside the normal printing area (200mm diameter circle). I've tried this with a few different STL files and most of them end up with this kind of excursion.
[Archive.zip](https://github.com/prusa3d/Slic3r/files/1045461/Archive.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Hugoz12, 2017-06-01:
Possible duplicate of #311

eanow, 2017-06-02:
Yes, appears to be the same issue. Extra wrinkle that the bug can generate moves that are actually outside the defined printable area.

bubnikv, 2017-06-02:
I believe this is a duplicate of
#311
#317

bubnikv, 2017-06-02:
Fixed with https://github.com/prusa3d/Slic3r/commit/b5f38dd23f413b06e87ebf005a658bf85bb70af6

closed 2017-06-02 by bubnikv