prusa3d/PrusaSlicer #324: Wipe Tower

arnaud-42, 2017-06-02:
### Version
1.35.1-prusa3d

### Operating system type + version
OSX

### Behavior
2 issues with the new wipe tower integration.

1: If you use a raft the wipe tower is generate only after last layer of raft ... (in the air)
You can see the bug here :
[https://imgur.com/a/otbpz](https://imgur.com/a/otbpz)

2: The second color on the tower is generate only after layer 2 (no raft for this test)
[https://imgur.com/a/c1TO6](https://imgur.com/a/c1TO6)

bubnikv, 2017-06-02:
Thanks for the report.

> 1: If you use a raft the wipe tower is generate only after last layer of raft ... (in the air)

The wipe tower is not compatible with the raft yet. I will need to work on that.

> 2: The second color on the tower is generate only after layer 2 (no raft for this test)

Would you please provide the configuration file and the platter as an AMF file?

Please note that only some support configuration is compatible with the wipe tower:
The non-soluble supports need to be printed with the active extruder (set the extruder ID to zero).
The soluble supports need to have their layers synchronized with the object layers.

BirkBinnard, 2017-06-03:
I'm unable to get the wipe tower to appear after slicing a part with dual extruders specified. I specified a skirt, no brim, no raft. Is there some trick to make it appear?

bubnikv, 2017-06-05:
As of now the wipe tower is generated for the "Single extruder multi material" option only.

[image]

and it needs to be enabled here:

[image]

Also please note that the current wipe tower implementation is very specific to the Prusa Multi Material setup. There are currently many hard coded assumptions, namely:
1) Nozzle diameter 0.4mm
2) Layer height 0.15 or 0.2
3) First layer height 0.2
4) Marlin firmware, relative extruder coordinates

I know these hard coded prerequisities are not nice, but we are supporting our products and we are improving them along the way.

BirkBinnard, 2017-06-05:
Thanks for this feedback. My problem was I had not checked the Single Extruder Multi Material box. Once I did this the wipe tower appeared OK.
[image]

bubnikv, 2017-06-06:
I have added checks for compatibility of the slic3r settings with the wipe tower into the Slic3r 1.35.2.

The raft issue is yet to be fixed.

bubnikv, 2017-06-08:
The raft support for the wipe tower has been implemented with
https://github.com/prusa3d/Slic3r/commit/89dcd3e8b11deda08d4f63cdd028c12c31c44a5b

closed 2017-06-08 by bubnikv