prusa3d/PrusaSlicer #325: "toggle" error and another

sarf2k4, 2017-06-03:
### Version
1.35.1

### Operating system type + version
Windows 10

### Behavior
I got this error once in a while, usually when I selects on a different printer profile. Furthermore, on an existing printer profile, it will automatically adds "(modified)" at the end of it. Some cases the slic3r would throw me an error like this http://share.pho.to/AiWfN and clicking "ok" would render it useless, kept appearing even more. 

Another thing is that sometimes the printer profile has a minor modification although it might not be serious but it leads to the above error sometimes. It occurs to me that slic3r kept adding the new variable introduced in 1.35.1 to the .ini file and then tags the printer profile has been modified 

At first I thought the libraries were clashing between the previous version, so I did delete the old one completely but here I am, got this error again

sarf2k4, 2017-06-03:
I have another one, basically I'm stuck and can't change to another printer profile http://share.pho.to/AiX4M

I am attaching the printer profile settings here as well
[printer profile.txt](https://github.com/prusa3d/Slic3r/files/1049582/printer.profile.txt)

What I did is to change the retract when the tool is disabled. 

On certain occasion, I would get the same error even without doing anything but to change to another printer profile. I tried removing the said printer profile above, the error still can come to me anyway. 

I had to use task manager to force close the slic3r and restart, only hoping there was no error but to no avail. I hope there will be a quick release to fix this issues

bubnikv, 2017-06-05:
I see a slight issue with the new Slic3r update and the "modified" flag. There are new parameters assigned to each extruder, and this Slic3r does not initialize the new parameters for extruder 2 and higher, therefore forcing the "modified" flag.

Would you please send me the complete content of your slic3r configuration?
C:\Users\yourname\AppData\Roaming\Slic3r
Thanks

sarf2k4, 2017-06-05:
As requested, providing every .ini files in a single zip
[Slic3r.zip](https://github.com/prusa3d/Slic3r/files/1051888/Slic3r.zip)

bubnikv, 2017-06-05:
I was unable to reproduce your issues with your config. The only issue I
was able to reproduce is the following message
[image: Inline image 1]
when switching from a multi-material printer to another printer. This is an
inconvenience only and I will try to fix it.

I see you have background processing enabled. The background processing is
generally not safe. If you disable background processing, will you still
get the issues?

Thanks

sarf2k4, 2017-06-05:
I was using the xyclone x1 multi mks based, switched to mk2 mm single ext.
and got the error straight. I was in the plater tab and background
processing still turned on, still giving me the same error when background
processing turned off

sarf2k4, 2017-06-05:
I think i can confirm that if the printer profile are 2 extruders, then
switching straight to the 4 extruder profile, would give me same error.

So to avoid this error, i had to switch to 1 extruder printer profile if i
were to go MM profile with 4 extruder from the other 2 extruder profile
that i have

bubnikv, 2017-06-06:
I have made a change to avoid the superfluous "Dirty" multi-extruder profile.
https://github.com/prusa3d/Slic3r/commit/35acd799cf097d1587527e02f8853e6900e9668e
Let's see whether it fixes your issues.

sarf2k4, 2017-06-06:
A quick question, since there are fixes to the slic3r to the current version which is 1.35.1, do I just download the file and overwrite the previous official release?

P.S
This is my first time D=

bubnikv, 2017-06-06:
In regard to the perl source code, it will likely work. In general, if you
cannot compile the whole project yourself, it is better wait for an
official build. I will do a bugfix build likely in the following days.

bubnikv, 2017-06-06:
Would you please check with the 1.35.2 release?

I have fixed the unnecessary "dirty printer configuration" notification due to the introduction of new extruder configuration parameters into Slic3r PE 1.35.1 (deretract_speed extruder_colour retract_before_wipe), so I hope it may have fixed your other issue as well.

sarf2k4, 2017-06-07:
I have confirmed that there are no more errors in 1.35.2, verified by
testing with 1.35.1 again

bubnikv, 2017-06-07:
Cool. Good news are always welcome. Thanks for your patience. Happily closing this issue.

closed 2017-06-07 by bubnikv