prusa3d/PrusaSlicer #327: Feature Request : Blip fans to full speed

jinie, 2017-06-04:
Some printers have a hard time starting the fans at low speeds, and it would be nice to have a feature to "blip the fans" to 100% to start them up, and then decrease fan speed to the desired speed, i.e. 25% immediately after.

bubnikv, 2017-06-05:
Do you have a proposal how to handle that feature at the G-code level?
It may be not quite straightforward to blip the fan without disturbing the path planning of the firmware.

jinie, 2017-06-05:
I was thinking something along the lines of
Whenever fan speed goes from 0 to xxx, insert a
M106 S255
M106 S<xxx>

It might not be easy to do, i haven’t checked.

/J

bubnikv, 2017-06-05:
> I was thinking something along the lines of
> Whenever fan speed goes from 0 to xxx, insert a
> M106 S255
> M106 S<xxx>

Please test it. What firmware are you using?
With Marlin, I am not sure whether the fan is synchronized with the planner
or not. Maybe the first M106 will be just ignored.

DrLex0, 2017-06-05:
In Sailfish, the fan is not synchronized. It will usually engage and stop sooner than would be expected from the code. This might be intentional because fans usually need quite a bit of time to spin up. I have found that inserting a ‘pipeline flush’ (G4 P0) before the fan command, does synchronize it.

bubnikv, 2017-06-05:
By the way, Marlin supports this feature:
FAN_KICKSTART_TIME
Will it not work for you?

bubnikv, 2017-06-05:
At least with Marlin, the fan speed is synchronized with the planner to some extent, so a set of M106 lines
M106 S255
M106 S<xxx>
may or may not work (likely will not work if the planner queue is nonempty).

It will work if you precede the M106 commands with a G4 command, but that would make the motion planner stutter, which is not a good solution either. The only good solution I can think of is for slic3r to interleave the M106 commands with the G1 commands, but that is a bit complex solution.

IMHO it is better to use the firmware feature if it is available.

bubnikv, 2017-06-06:
@jinie Does your firmware support the fan blip? As this feature will be uncomfortably firmware dependent, I would rather leave it to the firmware and not implement it into Slic3r.

jinie, 2017-06-06:
Looking at reprap.org, the description of M106 has an option of "blip time" 
``` 
M106: Fan On
Bnnn Blip time - fan will be run at full PWM for this number of seconds when started from standstill1
```
Which, as far as i can tell is supported on every variant.
The problem with setting it manually is that i will mess with the regular fan control, which i'd rather not.

bubnikv, 2017-06-06:
> Bnnn Blip time - fan will be run at full PWM for this number of seconds
when started from standstill1

This is certainly not supported by Marlin, at least not the version the
Prusa MK2 firmware is based upon.

1EyedJim, 2017-06-07:
Hello.  I would like to see this feature implemented into the software.  I love slic3r PE because it handles holes far better than S3D however S3D has the blip fan feature ;/  I need a 10% fan for ABS....

bubnikv, 2017-06-07:
@1EyedJim  What firmware do you run on?

1EyedJim, 2017-06-07:
Hello.  Running on the latest 3.0.11 version.

bubnikv, 2017-06-07:
@1EyedJim I will talk to Pavel, our firmware guy, to check the Marlin firmware feature.

1EyedJim, 2017-06-29:
@bubnikv Did you have a chat with your firmware guy?  Is this something that could be added?  Thanks

bubnikv, 2017-06-30:
The feature has always been present in Marlin firmware, so it is there. You may try to recompile our firmware with FAN_KICKSTART_TIME set, but please note that we did not test that yet.

// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100

bubnikv, 2017-10-03:
We will have the blip fan feature enabled and tested in our upcoming firmware for our Prusa i3 MK2 printer. We will not implement a fan blip feature into Slic3r as we believe that it is strongly firmware dependent.

Closing.

closed 2017-10-03 by bubnikv