prusa3d/PrusaSlicer #328: error in bed temperature when first layer is set to different temperature

ecl66, 2017-06-05:
### Version
1.34.1.24

### Operating system type + version
_What OS are you using, and state any version #s_
OS X El Capitan

When setting a different bed temperature for first and other layers, the first layer is not set to first temperature but to other layers' one.

bubnikv, 2017-06-05:
Would you please provide a config.ini file (go to menu file->export config), an STL (or rather an AMF file in case of a multi material print) and possibly the generated G-code? Thank you

ecl66, 2017-06-05:
here you are: 1st layer set at 75C, next layers at 80C. In gcode, bed temperature is set to 80C only.

thanks

bubnikv, 2017-06-05:
You did not attach any file.

ecl66, 2017-06-05:
Resent - files were attached (total 1.8MB). Cheers

here you are: 1st layer set at 75C, next layers at 80C. In gcode, bed temperature is set to 80C only.

thanks

bubnikv, 2017-06-06:
There is nothing attached. Please zip all files into a single zip, it should go through.

ecl66, 2017-06-06:
here are the files zipped. cheers

bubnikv, 2017-06-06:
Still nothing attached.

ecl66, 2017-06-06:
i am using Apple Mail - is that a problem?

ecl66, 2017-06-06:
> thread [2].
> Links:

trying again using webmail this time...

bubnikv, 2017-06-06:
Still nothing. I recommend to use the github web gui
https://github.com/prusa3d/Slic3r/issues/328

bubnikv, 2017-06-06:
Just click on the #328

bubnikv, 2017-06-13:
Reverted the "set_and_wait_temperatures" change  introduced in 1.35.1, as it was not in line with the latest @alexrj Slic3r and it caused confusion.

Please retest with the upcoming Slic3r 1.35.3. If that does not solve your issue, please reopen this issue.

closed 2017-06-13 by bubnikv