prusa3d/PrusaSlicer #329: Pressing Layer editing button crashes Slic3r

mroek, 2017-06-05:
On all versions after 1.34.1, Slic3er immediately crashes when pressing the "Layer editing" button. On 1.34.1 it works, on all subsequent versions it just crashes. What can I do to help diagnose the issue?

OS: Win 10 x64

bubnikv, 2017-06-05:
I believe it is a duplicate of https://github.com/prusa3d/Slic3r/issues/326
For now, you just need to set the maximum layer height for the extruder in the printer settings to a non zero value (75% of the nozzle diameter is a reasonable value).

closed 2017-06-05 by bubnikv

mroek, 2017-06-05:
Ok, sorry I didn't find #326 before posting. I can confirm your suggested workaround fixes the issue. Thanks.

bubnikv, 2017-06-05:
Good it works for you. Good news are always welcome.