prusa3d/PrusaSlicer #330: Crash when trying to enter layer editing mode in OS X

DrLex0, 2017-06-05:
### Version
1.35.1

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* Whenever I try to enter Layer Editing, Slic3r crashes. This has started in version 1.34.1.24. In 1.34.1, it worked.
* _Steps needed to reproduce the problem_
  * Load any STL file
  * Click the Layer Editing button
* _Expected Results_
  * It works
* _Actual Results_
  * It exits immediately, this is what I see in the console:

05/06/17 23:27:18,005 spindump[978]: Got xpc error message in libspindump client connection: Connection invalid
05/06/17 23:27:18,029 com.apple.xpc.launchd[1]: (com.prusa3d.slic3r/.569312[42763]) Service exited with abnormal code: 255
05/06/17 23:27:18,038 WindowServer[205]: Received display connect changed for display 0x4273402
05/06/17 23:27:18,048 WindowServer[205]: Found 22 modes for display 0x04273402 [22, 0]
05/06/17 23:27:18,059 WindowServer[205]: CGXMuxAcknowledge: Posting glitchless acknowledge
05/06/17 23:27:18,140 WindowServer[205]: Received display connect changed for display 0x4273402
05/06/17 23:27:18,140 WindowServer[205]: Found 1 modes for display 0x04273402 [1, 0]
05/06/17 23:27:18,141 WindowServer[205]: Received display connect changed for display 0x3f003f
05/06/17 23:27:18,141 WindowServer[205]: Found 1 modes for display 0x003f003f [1, 0]
05/06/17 23:27:18,141 WindowServer[205]: Received display connect changed for display 0x3f0040
05/06/17 23:27:18,141 WindowServer[205]: Found 1 modes for display 0x003f0040 [1, 0]

markmacpherson, 2017-06-06:
I see the same issue as well

pcsentinel, 2017-06-06:
Same issue in Win64 edition

bubnikv, 2017-06-06:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/329 https://github.com/prusa3d/Slic3r/issues/326 and it has been fixed already. Please confirm that setting the maximum layer height to 3/4 of the nozzle diamete at the printer extruder page fixes the issue.
I will leave this issue open until I release another build.

pcsentinel, 2017-06-06:
Yes, agree setting non zero value in max layer height stops crash, however hint still says set to 0 for default action

bubnikv, 2017-06-06:
A new build is on the way.

bubnikv, 2017-06-06:
Fixed with the Slic3r 1.35.2 - Prusa Edition release.

closed 2017-06-06 by bubnikv