prusa3d/PrusaSlicer #331: Illegal division by zero @ 3DScene.pm line 1479

realzeus, 2017-06-05:
### Version
1.35.1
(Slic3r-1.35.1-prusa3d-linux64-full-201706021450.AppImage)

### Operating system type + version
Debian 9 RC4

### Behavior
Slic3r exited immediately after I have clicked on some button, probably "Layer Editing".
Console has following output:
`Illegal division by zero at /tmp/.mount_QM1uWX/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/3DScene.pm line 1479.`

bubnikv, 2017-06-06:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/329 https://github.com/prusa3d/Slic3r/issues/326 and it has been fixed already. Please confirm that setting the maximum layer height to 3/4 of the nozzle diamete at the printer extruder page fixes the issue.
I will leave this issue open until I release another build.

bubnikv, 2017-06-06:
Fixed with the Slic3r 1.35.2 - Prusa Edition release.

closed 2017-06-06 by bubnikv