prusa3d/PrusaSlicer #332: Infill path extends into perimeter a tiny bit causing bumps on surface.

emilianomolina, 2017-06-06:
### Version
1.35.1

### Operating system type + version
Mac OS X 10.12.1

### Behavior
* When using any infill modes 'Triangles','Stars' and 'Cubic' on a 30mm wide cylinder the infill goes into the perimeter a little and causes bumps in the surface of the print.  Happens when the infill path starts and stops. It's visible in the 'Layers' view and the position of the infill going into the perimeter matches the position of the bumps in the print.
* To reproduce the problem slice the attached model with the perimeter settings listed above.

* _Expected Results_

Walls without bumps!
[image]

* _Actual Results_

Walls with bumps.

[image]

[image]

[image]

[image]

_Is this a new feature request?_

NO!

#### STL/Config (.ZIP) where problem occurs

[temperature test.stl.zip](https://github.com/prusa3d/Slic3r/files/1054793/temperature.test.stl.zip)

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1054798/config.ini.zip)

bubnikv, 2017-06-06:
The overlap of the infill lines with the inner perimeter loop is intentional and it is controlled by the "Perimeter infill overlap" percentage parameter.

The issue with the 'Triangles','Stars' and 'Cubic' pattern is a bit different. Slic3r Prusa Edition places these patterns by placing three trellis patterns rotated by 60 degrees one after the other without checking whether the other patterns overlap with the patterns already extruded. In most cases it is harmless, but with some softer materials like PET the excess material bulges through the outer shell wall.

When @alexrj integrated these patterns into his Slic3r, he disabled the perimeter following infill lines for the 2nd and 3rd pass. I did not do that as it reduces the infill - perimeter bonding. I hope to develop a smarter solution by explicitly testing for the overlap.

For now, if the multiple extrusions are problematic, the grid pattern produces less of this unwanted effect and the honeycomb is free of it.

emilianomolina, 2017-06-06:
Thank you for your prompt reply!

I know about the "Perimeter infill overlap" setting, I've actually set it to 0 to see if it helped.  Thanks for highlighting it though!  it's an important one.

Please understand that the issue I'm seeing is not at every point where the infill pattern touches the perimeter.  It's only when the path starts and stops that there is an issue. If you look at the second screenshot I've highlighted the exact spots. 

Your answer kind of implies that I didn't get that point across the first time.

bubnikv, 2017-06-06:
>Please understand that the issue I'm seeing is not at every point where the infill pattern touches the perimeter. 

I understand now. You may be right, though there is a compromise between a good boding of the infill line with the inner perimeter against the effect you are experiencing. Geometrically the extruder pushes too far, but in regard to the volume extruded it should fit. 

Whether it will bulge into the perimeter depends on the material viscosity, on the hardness of the outer shell (PLA is rigid, PET and ABS are less rigid), and on the lag of extruder and the changing extruder pressure effects (slowing the XY movement down usually leads to excessive extrusion if printed fast).

bubnikv, 2017-06-06:
By the way, what filament material is it and what printer? Is it a direct drive extruder or a bowden extruder?

emilianomolina, 2017-06-06:
It's a Prusa Mk2S with PLA.

My first 3d printer!

If you look carefully at the layer paths I've included at the top you'll see that it goes a little into the perimeter in only the places where the photos show the bulges.

Setting 'outside perimeter' first doesn't help and the fan is on.

I'm not an expert (and the printer for like three weeks), but my thought was that it was re-melting the whole area when the filament is applied.

If you look at the pictures you'll see that some of the bulges are a lot more prominent than the others.  What I thought may be happening is that when the filament is extruded into places where there has already been some infill done, we are trying to put plastic into almost the same spot for the third time. If it's also up against the perimeter it may be melting the perimeter a bit and with the added pressure it may be enough to move the plastic out a bit.

But really, I'm like a monkey learning to drive a car here, I have some ideas but they're not based on much experience!  I just thought that those paths going beyond the perimeter of the other infill was an issue.

ceryen, 2017-06-06:
In your config.ini, I see that you have "Seam position" set to "Random".  Strangely, that doesn't seem to be causing the start/stop point of each layer to be distributed uniformly across the surface of the model.  Instead, "Random" seems to be causing the start/stop point of each layer to be distributed *only* at the points where infill intersects the perimeters.

If you set "Seam position" to "Aligned" or "Rear", I bet you would see the blobbing aligned in a straight vertical line.  If that's what happens, then what you're seeing is not blobbing caused by infill, but rather blobbing caused by the start/stop point of each layer.  You would then want to focus your attention on reducing blobbing caused by the start/stop point of each layer.

I don't think it's intentional for "Seam position" set to "Random" to only distribute start/stop points to only the points where infill intersects the perimeters, so that may be an actual issue.

ceryen, 2017-06-06:
Actually, I don't think "Random" is causing the start/stop points to be distributed along the points where infill intersects the perimeters.  I think "Random" is causing the start/stop points to fall within a few vertical lines across the surface of the model, and it only *looks* like the vertical lines are aligned to where infill intersects the perimeters.

I still don't think it's intentional for "Random" to only distribute start/stop points along a few vertical lines across the surface of the print, so I think that may still be an issue.

emilianomolina, 2017-06-07:
I will print the same model with "aligned" and "rear" and see what we get!

fiveangle, 2017-06-07:
@bubnikv - One thing I have also noticed with LIN_ADVANCE available in Marlin 1.1 is that toolpath stop-start and Z-change scaring is reduced, presumably by pre-calculating pressure so it is minimized during these transitions.

Regarding this issue, your theory seems reasonable, but if the preview correctly depicts the tool path, it does appear that the algorithm could be improved.  Have you tested @alexrj's treatment vs. yours to confirm lack of infill-perimeter bonding ?  It may be worth revisiting to devise some test that can prove-disprove your working theory.  I've certainly noticed these types of visual artifacts as well.

BTW- The advancements Slic3rPE is making is nice to see.  Good work !

emilianomolina, 2017-06-08:
@ceryen , you are right! When I set the seam position to rear all the blobs went to the back.  I wasn't able to see anything on the perimeter of the print where the infill joins to it.

Thank you for your help.

bubnikv, 2017-06-08:
> In your config.ini, I see that you have "Seam position" set to "Random". Strangely, that doesn't seem to be causing the start/stop point of each layer to be distributed uniformly across the surface [...]

Currently Slic3r puts the random seams at the end points of line segments and the line segments are created by cutting the triangles of the input geometry with a slicing plane. So the effect you described seems to be a coincidence of the input geometry discretization being aligned with the infill lines. 

@emilianomolina So it seems your issue is really too high expectations of the FDM print technology :-) The seams cannot be fully avoided, but they could be reduced to some extent, for example by the linear advance of the printer firmware as @fiveangle mentioned. So can we close this issue?

@fiveangle I may look into our firmware to try to port the linear advance once this Slic3r release is finalized.

bubnikv, 2017-06-08:
@fiveangle 
> Regarding this issue, your theory seems reasonable, but if the preview correctly depicts the tool path, it does appear that the algorithm could be improved. Have you tested @alexrj's treatment vs. [...]

I would like to properly test for the overlap of the infill lines following the perimeter, and trim them where there is an excess material extruded. The implementation by @alexrj is too simplistic to me. This has a pretty high priority to me together with the support issues.

emilianomolina, 2017-06-08:
@bubnikv you can close the issue if you believe that the problem with the tool path does not warrant keeping the issue open.

I've spent some more time experimenting with the extruder temperature to manage the pressure a bit better and have come up with a result that I am more than happy with. 

[image]

[image]

These shots are taken through a magnifying glass, the piece is only 30mm across. The seam is very hard to see unless the light is jus right and as you can see with the second image it hardly affects the external dimensions now if at all.

Thank you all for your explanations and working through the issue.

bubnikv, 2020-11-20:
PrusaSlicer 2.3.0-alpha4 controls anchoring of the infill lines to the innermost perimeter better:
1) The anchoring lines never overlap
2) The open ends of infill lines are anchored as well.
I believe it should improve this particular print.
Closing.

closed 2020-11-20 by bubnikv